FreeCAD/FreeCAD #21111: Single letter variables do not enable the Ok button

leoheck, 2025-05-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I want to create quick-to-use variables... like a single letter.
But some letters seem not to be allowed. Why? Is this a bug?

https://github.com/user-attachments/assets/6c905158-ddb7-4c1e-9601-72df8ab23985

Also, the tooltip of the `Name` field should show the rules for creating the name.
Other fields should also have tooltips.

Or perhaps it could have an initial text/guide, so the rules are written explicitly above the fields.

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.41464 (Git)
Build date: 2025/04/28 16:20:16
Build type: Release
Branch: main
Hash: 8d6547d519ffb331cfba054b4db2750b3e71339b
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.74.0
  * Curves 0.6.61
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * Curves.backup1746105636.3777354 0.6.60 (Disabled)
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.2.0
  * lattice2 1.0.0
  * kicadStepUpMod 11.6.9
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-03:
@pieterhijma FYI

leoheck, 2025-05-03:
@Katt-Kushol this also looks like a different window.

Why a different window layout?

belinski-roman, 2025-05-04:
The reservation of some one-letter numbers seems to be related to the name of the units of measurement, if I'm not confusing anything

pieterhijma, 2025-05-08:
> [@Katt-Kushol](https://github.com/Katt-Kushol) this also looks like a different window.
> 
> Why a different window layout?

Because this Add Property dialog gives more feedback when an identifier doesn't match. There is one for VarSets and there is one for the Property View. The plan is to end up with only one.

pieterhijma, 2025-05-08:
> The reservation of some one-letter numbers seems to be related to the name of the units of measurement, if I'm not confusing anything

That is indeed the case.

leoheck, 2025-05-08:
> That is indeed the case.

Yeah, that's obvious. But why are variables not in a different container or something that prevents this collision?

This window, as it is now, is better than before, where any forbidden character would result in an annoying pop-up. 

But there are some UI/UX issues here.

Characters should be checked only when clicking the Ok button.
Otherwise user cannot use the field to "think" while typing, and then the computer becomes an untouched interface where the user can only put valid things, and it has to "think" in another place, like a paper, or a text editor.

Also, if there are restricted rules for naming variables and for filling the fields, these should be written inside the window clearly a a description close to the fields, or at least a link to the formatting could be given at that time, so the user can learn while using this feature (like the image that was atted to the PardDesign Hole)

pieterhijma, 2025-05-14:
> > That is indeed the case.
> 
> Yeah, that's obvious. But why are variables not in a different container or something that prevents this collision?

I don't know exactly. I think someone simply defined the policy that names of properties should not collide with units, perhaps for a good reason. There is also the fact that property names should be accessible in Python, so you can imagine that it is pretty complicated.

> This window, as it is now, is better than before, where any forbidden character would result in an annoying pop-up.

That's good to hear!
 
> But there are some UI/UX issues here.
>
> Characters should be checked only when clicking the Ok button. Otherwise user cannot use the field to "think" while typing, and then the computer becomes an [...]

That is in contrast to https://github.com/FreeCAD/FreeCAD/issues/17474. 
 
> Also, if there are restricted rules for naming variables and for filling the fields, these should be written inside the window clearly a a description close to the fields, or at least a link to the [...]

I can see that, but I think this is such a much used dialog that this is a bit overkill. I would recommend to create a separate issue on that.

pieterhijma, 2025-05-14:
@maxwxyz and others. I would recommend to close this issue because it is actually a feature and was requested in #17474. For more specific issues, I would recommend to create new ones for those.

closed 2025-05-14 by maxwxyz

maxwxyz, 2025-06-06:
@pieterhijma did you see https://github.com/wwmayer/FreeCAD/commit/ce523e64b5535c68c5cc7bfdf29ce4de6f65123b and the two commits before that?