FreeCAD/FreeCAD #21114: TechDraw: AngleDimension for small angle

mvxe, 2025-05-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Using TechDraw AngleDimension ("Insert Angle Dimension) does not work for almost parallel edges, so one cannot mark very small angles. It Fails with an error "Can not make 2D dimension from selection".

If I do make an AngleDimension on two lines with a larger angle, and then change the referenced object to a small angle, I get an exception:
`<Exception> Can not make angle from parallel edges`

The limit seems to be 0.028 deg. Looking at the code: the above exception happens when calling `DrawUtil::fpCompare(fabs(leg0Dir.Dot(leg1Dir)), 1.0))`
And fpCompare tolerance defaults to [epsilon float](std::numeric_limits<float>::epsilon()):
```
static bool fpCompare(const double& d1, const double& d2,
                          double tolerance = std::numeric_limits<float>::epsilon());
```
If you run the numbers the resulting tolerance here is indeed about 0.028 deg. I am not sure if the initial AngleDimension uses the same tolerance or fails in a different way, so there might need to be two changes to the code to fix this.

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (XFCE/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 115/153.365
Installed mods: 
  * 3DfindIT 1.2.0
  * Manipulator 1.5.7
  * Assembly4 0.50.17
  * pyrate
  * A2plus 0.4.68
  * Part-o-magic 1.1.0
  * slic3r-tools
  * fasteners 0.5.37
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-05-03:
Usually we use EWTOLERANCE (0.0001) in fpCompare instead of the default tolerance.

mvxe, 2025-05-03:
Well here the tolerance is unspecified:
https://github.com/FreeCAD/FreeCAD/blob/d427162d976b5724c4bd9a985fc16fb898dc4697/src/Mod/TechDraw/App/DrawViewDimension.cpp#L1260

Note that, if you used EWTOLERANCE (0.0001) for angle comparison (i.e. what is compared is the 1-cos(angle/180*pi)<tolerance ), then the limit would be even worse: 0.81 deg.

mvxe, 2025-05-03:
On a second check, the limit to create is currently indeed 0.81 deg. To modify it afterwards it is 0.028 deg
The first limit is because that, even if you explicitly do the angle tool, it still checks if the lines are parallel via:
https://github.com/FreeCAD/FreeCAD/blob/d427162d976b5724c4bd9a985fc16fb898dc4697/src/Mod/TechDraw/Gui/DimensionValidators.cpp#L544
Then the other check is when drawing, see above comment.

The EWTOLERANCE (0.0001) works for distances (i.e. mm units), but it becomes massive in terms of the dot product. For the tolerance to be 0.0001 deg, modifying the above two lines to:
`if (DrawUtil::fpCompare(fabs(leg0Dir.Dot(leg1Dir)), 1.0, 1-cos(EWTOLERANCE/180*M_PI))) {`
and
`if (DU::fpCompare(dot, 1.0, 1-cos(EWTOLERANCE/180*M_PI))) {`
Makes it work for angles bigger than 0.0001 deg. The problem with this modification is that this also changes the behavior for the "Insert Dimensions".
Why is the code even verifying when one tries to explicitly insert an angle dimension if the lines are parallel? Really what it should do imo is insert  a 0 degree dimension. And why is there even an angle tolerance check in DrawViewDimension, why not just leave the angle at full double precision?

Anyway I have my workaround, but perhaps this behavior with angles in TechDraw need rethinking.

commit 4dd23134 (2025-05-26): [TD] fix small angle dims (fix #21114) (#21208)

* [TD]fix handling of very small angle dimensions

- also remove test for too small format specifier

* [TD]lint messages

* [TD]apply review comments.
closed 2025-05-26 by kadet1090