FreeCAD/FreeCAD #21115: TechDraw: Hatch and GeometericHatch don't save the last used settings

semhustej, 2025-05-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature**

**Problem description**
When user invokes `TechDraw_Hatch` or `TechDraw_GemotericHatch` repeatedly, he/she needs to do the whole setup (set pattern file, set scale, color line, rotation, etc.) all over with each start of the tool. 

**Proposed solution**
Save last used settings of `TechDraw_Hatch` or `TechDraw_GemotericHatch` so that when user invokes the tool second time, the last used settings are loaded automatically.
It is more probable, that user will utilize the last used setting than the default one. Saving hatch settings would save user a lot of time, because the user would not need to set up the hatch each time it us used.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * BIM021 2021.12.0
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

TechDraw

### Anything else?

Draft_Hatch remembers hatch settings.
Draft_Hatch remembers hatch settings even after FreeCAD is restarted. I think it would be better to re-set these setting after FreeCAD is restarted. But on the other hand it would be nice for both Draft and TechDraw hatches to behave in the same way, for the sake of user interface consistency.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

forbes-0023, 2026-02-19:
I can work on this if nobody else is

semhustej, 2026-02-19:
I thought there could be a `Reload default` button next to the path selection button. It would definitely have to have a tooltip. I am not sure if this needs a DWG discussion before implemented. I don't recall any other such functionality in other places in FreeCAD.

[image]

maxwxyz, 2026-11-07:
Part of this is implemented in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/a4a35b03798323b1a7cbe5ca7bd2bc164e54764d

Its replacement Hatch Face tool remembers the last settings. This needs the new tool from https://github.com/AstoCAD/FreeCAD/commit/3dc963b527352bae5cd01fa8539b21badda62293, it is not a standalone change to the two upstream commands.