FreeCAD/FreeCAD #21116: Delete broken link core dumped

xobes, 2025-05-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

File attached: `broken_link_core_dump.tgz` 

1. Open `light_cube.FCStd` 
2. select and delete `side_leg002` via the tree view
3. Observe crash:
```...
Observer 0x7ffed85916e8 already detached
Observer 0x7ffed85916b8 already detached
<PropertyLinks> PropertyLinks.cpp(3493): reloading partial document '/home/xobes/projects/3d/freecad/ideas/light_cube/side.FCStd' due to object side_leg
The property Reference1 of Joint light_cube#Joint008 is bad.FreeCAD: /home/xobes/projects/freecad/freecad-source/src/3rdParty/OndselSolver/OndselSolver/LimitIJ.cpp:112: virtual void MbD::LimitIJ::fillqsudot(FColDsptr): Assertion `false' failed.
Aborted (core dumped)
```

Workaround: If either link `micro_servo_SG92R` or `disc` are deleted first, then `side_leg002` can be deleted without crashing.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (fvwm3/xcb)
Architecture: x86_64
Version: 1.1.0dev.41491 (Git)
Build date: 2025/05/02 22:58:22
Build type: Unknown
Branch: development
Hash: d427162d976b5724c4bd9a985fc16fb898dc4697
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/86.4394/1
Installed mods: 
  * MakerWorkbench 1.0.1
  * fasteners 0.5.38
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Anything else?

[broken_link_core_dump.zip](https://github.com/user-attachments/files/20023927/broken_link_core_dump.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-05:
Is this a link or assembly ticket ?

luzpaz, 2025-05-15:
No core dump, just the following output:

```
22:25:22  pyException: <string>(1)<class 'ModuleNotFoundError'>: No module named 'lattice2PolarArray2'
22:25:22  pyException: <string>(1)<class 'ModuleNotFoundError'>: No module named 'lattice2PopulateCopies'
22:25:22  pyException: <string>(1)<class 'ModuleNotFoundError'>: No module named 'FuseCompound2'
22:25:22  pyException: <string>(1)<class 'ModuleNotFoundError'>: No module named 'lattice2PolarArray2'
22:25:22  pyException: <string>(1)<class 'ModuleNotFoundError'>: No module named 'lattice2PopulateCopies'
22:25:22  pyException: <string>(1)<class 'ModuleNotFoundError'>: No module named 'FuseCompound2'
22:25:22  <PropertyLinks> PropertyLinks.cpp(3492): reloading partial document '/home/foo/Downloads/broken_link_core_dump/side.FCStd' due to object side_leg
22:25:40  The property Reference1 of Joint light_cube#Joint008 is bad.
```

```
FreeCAD_weekly-builds-41694-conda-Linux-x86_64-py311.AppImage
```

So weird, but I can't access the About menu in the latest appimage.

luzpaz, 2025-05-15:
@xobes can you try to reproduce in Safe Mode[^1] ? 

[^1]: https://wiki.freecad.org/Std_RestartInSafeMode

luzpaz, 2025-05-17:
@xobes ?

xobes, 2025-05-17:
I can.

luzpaz, 2025-05-19:
@xobes Can you test on the latest dev build ? Please paste your [About](https://wiki.freecad.org/About) info when you do, TIA

xobes, 2025-05-19:
Whoops, I meant "I can [try]" -- I have not tried yet.  I am working from a git repo that I'm keeping in sync with the main branch every time I compile.  I'll try again and post when I have tried.
I will use a fresh build of the latest source and I will (also) try in safe mode.  I will provide my build info and results when I have done this.

xobes, 2025-05-23:
Sorry for the delay:

```
 ./build/bin/FreeCAD --safe-mode
FreeCAD 1.1.0, Libs: 1.1.0devR41837 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

FreeCAD is running in _SAFE_MODE_.
Safe mode temporarily disables your configurations and addons. Restart the application to exit safe mode.

Migrating Start Workbench to Start command... done.
<string>(1)<class 'ModuleNotFoundError'>: No module named 'lattice2PolarArray2'
<string>(1)<class 'ModuleNotFoundError'>: No module named 'lattice2PopulateCopies'
<string>(1)<class 'ModuleNotFoundError'>: No module named 'FuseCompound2'
<string>(1)<class 'ModuleNotFoundError'>: No module named 'lattice2PolarArray2'
<string>(1)<class 'ModuleNotFoundError'>: No module named 'lattice2PopulateCopies'
<string>(1)<class 'ModuleNotFoundError'>: No module named 'FuseCompound2'
<PropertyLinks> PropertyLinks.cpp(3493): reloading partial document '/home/xobes/temp/2025/05/broken_link_core_dump/side.FCStd' due to object side_leg
The property Reference1 of Joint light_cube#Joint008 is bad.FreeCAD: /home/xobes/projects/freecad/freecad-source/src/3rdParty/OndselSolver/OndselSolver/LimitIJ.cpp:112: virtual void MbD::LimitIJ::fillqsudot(FColDsptr): Assertion `false' failed.
Aborted (core dumped)
```

OS: Ubuntu 24.04.2 LTS (fvwm3/xcb)
Architecture: x86_64
Version: 1.1.0dev.41837 (Git)
Build date: 2025/05/23 13:10:01
Build type: Unknown
Branch: development
Hash: a969f31573c23d0aa956e87a1528d4cc393468ba
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/86.4394/1

luzpaz, 2025-09-26:
@xobes you built FC from source ?

xobes, 2025-09-26:
Yes, let me know if you need me to try again, that was a while ago.

FlachyJoe, 2025-11-06:
@xobes  #25126 fix a bug I've found with link deletion, can you try it?

xobes, 2025-11-23:
I had the following results:

#### Linux:
 - Works as expected, did not crash.

#### Windows (FreeCAD_weekly-2025.11.20-Windows-x86_64-py311):
 - freecad.exe from weekly build does not run much past the splash screen -- could not test..

luzpaz, 2025-12-01:
@xobes we have progress on Linux at least. 

Any chance you can test the latest weekly (2025.11.26) on windows. And output the console + include the full [About](https://wiki.freecad.org/About) info? Thx!

luzpaz, 2025-12-25:
@FEA-eng can you test on windows ?

FEA-eng, 2026-08-27:
Still no crash on Windows

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260825 (Git shallow)
Build date: 2026/08/25 17:50:34
Build type: Release
Branch: (HEAD detached at 964fce2ed)
Hash: 964fce2edc9a7e1905c2fd3fcd6e7be90afc8375
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```