FreeCAD/FreeCAD #21117: Unable to remove old cosmetic lines

dannystaple, 2025-05-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I've opened a Techdraw created in a version of FreeCAD prior to 1.0.0, selected a cosmetic line.

I expected to find the Erase Cosmetic Line tool - but this is gone as per https://wiki.freecad.org/TechDraw_CosmeticEraser.

I attempted delete, but this deleted the whole view from the page.

My workaround currently is to hide old lines, as I don't see a way to remove them.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * ExplodedAssembly
  * fasteners 0.5.38
  * kicadStepUpMod 11.6.9
  * Manipulator 1.6.0
  * Telemetry 1.0.1
```

### Subproject(s) affected?

None

### Anything else?

I see no current open issue relating to cosmetic lines.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-04:
Can you test on 1.1-dev[^1] ?

[^1]: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets

Syres916, 2025-05-06:
I cannot confirm with either the 1.0 AppImage or the up-to date compiled 1.0 Branch (i.e. with all current backported fixes). both the `Delete` key and the menu entry _TechDraw>Remove Cosmetic Object_ work correctly on a 0.21.x created file. We ideally need a Windows user to see if it's platform specific.

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-Ribbon 1.8.1.2 (Disabled)
  * btl 0.9.9
  * FreeCAD-themes 2025.1.7
  * Silk 0.1.6
  * sheetmetal 0.7.22
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.30.3
  * ChangeAppFont
  * Quetzal 1.4.4
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * Curves 0.6.61
  * QetWireManager
  * A2plus 0.4.68
  * ThreadProfile 1.94.0
```

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39275 (Git)
Build type: Release
Branch: releases/FreeCAD-1-0
Hash: deb68996f357bf02161b449036042d5e39ff0867
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-Ribbon 1.8.1.2 (Disabled)
  * btl 0.9.9
  * FreeCAD-themes 2025.1.7
  * Silk 0.1.6
  * sheetmetal 0.7.22
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.30.3
  * ChangeAppFont
  * Quetzal 1.4.4
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * Curves 0.6.61
  * QetWireManager
  * A2plus 0.4.68
  * ThreadProfile 1.94.0
```

dannystaple, 2025-05-07:
A further observation here (While I wait for the nightly to decompress) is that I *can* remove things with menu->techdraw->"remove cosmetic object". But not with the delete button.

WandererFan, 2025-05-13:
The del key works to delete cosmetic lines with these 2 versions:
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41464 (Git) AppImage
Build date: 2025/04/28 16:20:16
Build type: Release
Branch: main
Hash: 8d6547d519ffb331cfba054b4db2750b3e71339b
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/118.229/1.19792
Installed mods: 
  * fasteners 0.5.21

OS: Linux Mint 21.2 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41632 (Git)
Build date: 2025/05/09 19:27:21
Build type: Unknown
Branch: CosmeticEdgeDelete21117
Hash: ce368cb7a0dd6a3f57e5cabc3e7745c5669683a9
Python 3.10.12, Qt 5.15.2, Coin 4.0.0, Vtk 7.1.1, boost 1_74, Eigen3 3.4.0, PySide 5.15.2
shiboken 5.15.2, SMESH 7.7.1.0, xerces-c 3.2.3, OCC 7.8.0
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 115/141.585/1
Installed mods: 
  * fasteners 0.5.21

luzpaz, 2025-05-13:
@dannystaple can you use the most up to date version of 1.1dev to confirm ?

dannystaple, 2025-05-13:
I've been able to confirm something interesting here. In some of the files I have with techdraw - Del works and removes the cosmetic line. In other older files, it doesn't. It seems to be related to the age of the file. 
That is reproducible in the nightly and the release - new techdraw cosmetic lines have no issue. 
I'll find a place to host a problematic file so you can try this file.

luzpaz, 2025-05-13:
@dannystaple can you zip the file and drag/drop it into the github textbox in order to attach it ?

dannystaple, 2025-05-13:
(ahh - didn't know it would accept zips - that might be better than a google docs link... coming up)

[fitting-distance-sensor.zip](https://github.com/user-attachments/files/20189195/fitting-distance-sensor.zip)

I had tested this in `Version: 1.1.0dev.41365 (Git)`. You may need the exploded assembly workbench to fully interact with that file.

luzpaz, 2025-05-13:
41365 is not the latest dev build JFYI,

WandererFan, 2025-05-16:
Is it possible that that deleting a cosmetic line doesn't work in a clip group, but does work outside a clip group?  I don't use clip groups, but SEnsorDrawingPage has them (DEL removes the view) and the other page does not (DEL removes selected cosmetic line).

dannystaple, 2025-05-19:
Ahh, I'd not considered the clip group, that is a likely factor here. When I find a moment, I can try adding a clip group to a fresh simpler drawing with cosmetic line and see if I can repo that way.

dannystaple, 2025-05-19:
Attempted repro case:
- Open FreeCAD
- New Document, new part design body, sketch simple rect with circle 
- Pad this
- Enter tech draw
- Default page
- Insert a view (I chose top)
- Create a clip group, select the top view
- Add circle centre lines
- Attempt to delete a centre line.
Yes - the view is removed from the clip group, and not the line from the view.

WandererFan, 2025-08-12:
@maxwxyz this should go into the next release.  Does that make it a "blocker"?

closed 2025-08-25 by chennes