FreeCAD/FreeCAD #21120: Pipe/sweep on ellipse arc section freezes FreeCAD forever

ElementW, 2025-05-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Yet another https://github.com/FreeCAD/FreeCAD/labels/3rd%20party%3A%20OCC issue.

Trying to apply a pipe/sweep on a section'd spine containing an arc section just freezes FreeCAD if the mode is Standard or Frenet. Fixed is not affected.

[model.FCStd.zip](https://github.com/user-attachments/files/20025367/model.FCStd.zip). Just trigger a recompute (of the pipe).

### Full version info

```shell
OS: Void Linux (KDE/wayland)
Architecture: x86_64
Version: 1.1.0dev.41464 (Git) AppImage
Build date: 2025/04/28 16:20:16
Build type: Release
Branch: main
Hash: 8d6547d519ffb331cfba054b4db2750b3e71339b
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * fasteners 0.5.31
  * OpenTheme 2024.9.1
  * FreeCAD-Ribbon 1.8.1.2
  * SearchBar 1.0.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-04:
@pieterhijma FYI

ElementW, 2025-05-04:
Oh it's *worse*.

* [Sectioned B-spline](https://github.com/user-attachments/files/20027018/brokenBsplineSection.FCStd.zip): hang
* [Sectioned circle arc](https://github.com/user-attachments/files/20027019/brokenCircleArcSection.FCStd.zip): `BRep_API: command not done`
* [Sectioned hyperbola arc](https://github.com/user-attachments/files/20027020/brokenHyperbolaArcSection.FCStd.zip): hang
* [Sectioned parabola arc](https://github.com/user-attachments/files/20027022/brokenParabolaArcSection.FCStd.zip): hang

pieterhijma, 2025-05-14:
I can confirm. There seems to be an endless loop.

<details>
<summary>Version info</summary>
<pre>
OS: Arch Linux (tty)
Architecture: x86_64
Version: 1.1.0dev.41697 (Git)
Build date: 2025/05/14 07:36:42
Build type: Unknown
Branch: main
Hash: 9d6f1ad37c1d97bcf669272c3124ce6045691b5b
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 6.9.0
shiboken 6.9.0, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 8.0.0.rc1-7aa85582ad
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
  * Curves 0.6.61
  * freecad.gears 1.3.0
</pre>
</details>

<details>
<summary>stacktrace 1</summary>
<pre>
#0  0x00007fff7597a49d in BSplCLib::MultForm (Mults=..., FromK1=1, ToK2=2507) at /home/pieter/current/occt-liaison/occt-main/src/FoundationClasses/TKMath/BSplCLib/BSplCLib.cxx:686
#1  0x00007fff7597a531 in BSplCLib::KnotAnalysis (Degree=1, Periodic=false, CKnots=..., CMults=..., KnotForm=@0x55555ed15cf4: GeomAbs_NonUniform, MaxKnotMult=@0x7fffffff6034: 0) at /home/pieter/current/occt-liaison/occt-main/src/FoundationClasses/TKMath/BSplCLib/BSplCLib.cxx:708
#2  0x00007fff76f21b8f in Geom_BSplineSurface::UpdateVKnots (this=0x55555ed15ce0) at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKG3d/Geom/Geom_BSplineSurface.cxx:1154
#3  0x00007fff76f1c771 in Geom_BSplineSurface::Geom_BSplineSurface (this=0x55555ed15ce0, Poles=..., Weights=..., UKnots=..., VKnots=..., UMults=..., VMults=..., UDegree=6, VDegree=1, UPeriodic=true, VPeriodic=false) at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKG3d/Geom/Geom_BSplineSurface.cxx:278
#4  0x00007fff76f1b94f in Geom_BSplineSurface::Copy (this=0x55555eb5bca0) at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKG3d/Geom/Geom_BSplineSurface.cxx:147
#5  0x00007fff76f39998 in Geom_Geometry::Transformed (this=0x55555eb5bca0, T=...) at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKG3d/Geom/Geom_Geometry.cxx:136
#6  0x00007fff7674ce1b in BRepAdaptor_Surface::UTrim (this=0x55555eb2d150, First=0, Last=6.2831853071795862, Tol=1.0000000000000001e-09) at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKBRep/BRepAdaptor/BRepAdaptor_Surface.cxx:146
#7  0x00007fff76a56f79 in Adaptor3d_HSurfaceTool::UTrim (theSurf=..., theFirst=0, theLast=6.2831853071795862, theTol=1.0000000000000001e-09) at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKG3d/Adaptor3d/Adaptor3d_HSurfaceTool.hxx:97
#8  0x00007fff76a4ec6f in IntCurveSurface_HInter::InternalPerform (this=0x7fffffff67f0, curve=..., polygon=..., surface=..., U1=0, V1=2, U2=6.2831853071795862, V2=3) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKGeomAlgo/IntCurveSurface/IntCurveSurface_Inter.gxx:875
#9  0x00007fff76a4c0f0 in IntCurveSurface_HInter::Perform (this=0x7fffffff67f0, curve=..., surface=..., U1=0, V1=2, U2=6.2831853071795862, V2=3) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKGeomAlgo/IntCurveSurface/IntCurveSurface_Inter.gxx:445
#10 0x00007fff76a4b9e9 in IntCurveSurface_HInter::Perform (this=0x7fffffff67f0, curve=..., surface=...) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKGeomAlgo/IntCurveSurface/IntCurveSurface_Inter.gxx:357
#11 0x00007fff783b5e9a in IntTools_BeanFaceIntersector::ComputeAroundExactIntersection (this=0x7fffffff6aa0) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKBO/IntTools/IntTools_BeanFaceIntersector.cxx:570
#12 0x00007fff783b4b46 in IntTools_BeanFaceIntersector::Perform (this=0x7fffffff6aa0) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKBO/IntTools/IntTools_BeanFaceIntersector.cxx:348
#13 0x00007fff783e0bcf in IntTools_EdgeFace::Perform (this=0x55555add21f0) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKBO/IntTools/IntTools_EdgeFace.cxx:552
#14 0x00007fff784b8233 in BOPAlgo_EdgeFace::Perform (this=0x55555add21a0) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKBO/BOPAlgo/BOPAlgo_PaveFiller_5.cxx:130
#15 0x00007fff784bab59 in BOPTools_Parallel::ContextFunctor2<NCollection_DynamicArray<BOPAlgo_EdgeFace>, IntTools_Context>::operator() (this=0x7fffffff7900, theThreadIndex=0, theIndex=0) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKBO/BOPTools/BOPTools_Parallel.hxx:141
#16 0x00007fff784baa2e in OSD_ThreadPool::Job<BOPTools_Parallel::ContextFunctor2<NCollection_DynamicArray<BOPAlgo_EdgeFace>, IntTools_Context> >::Perform (this=0x7fffffff78a0, theThreadIndex=0) at /home/pieter/current/occt-liaison/occt-main/src/FoundationClasses/TKernel/OSD/OSD_ThreadPool.hxx:322
#17 0x00007fff754e2a27 in OSD_ThreadPool::performJob (theFailure=..., theJob=0x7fffffff78a0, theThreadIndex=0) at /home/pieter/current/occt-liaison/occt-main/src/FoundationClasses/TKernel/OSD/OSD_ThreadPool.cxx:263
#18 0x00007fff754e1b92 in OSD_ThreadPool::EnumeratedThread::WakeUp (this=0x7fffffff7958, theJob=0x7fffffff78a0, theToCatchFpe=false) at /home/pieter/current/occt-liaison/occt-main/src/FoundationClasses/TKernel/OSD/OSD_ThreadPool.cxx:48
[... cut ...]
#30 0x00007fff772c6378 in BRepBuilderAPI_MakeShape::Shape (this=0x7fffffff96a0) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKTopAlgo/BRepBuilderAPI/BRepBuilderAPI_MakeShape.cxx:35
#31 0x00007fff6b70e7fa in PartDesign::Pipe::execute (this=0x55555e688430) at /home/pieter/current/occt-liaison/freecad-occt-main/src/Mod/PartDesign/App/FeaturePipe.cpp:325
</pre>
</details>

Somewhere in the functions in the stacktrace there is function that doesn't converge. I'll add it to the OCCT Liaison project.

GeeBee2, 2026-11-06:
The recompute works now.

It is just slow, something like 10s for me.

It freezes however if I want to use check geometry.

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 27.1.0dev.20261006 (Git shallow)
Build date: 2026/10/06 21:00:40
Build type: Release
Branch: (HEAD detached at 3424c4106)
Hash: 3424c410627263394f701aeec5b61d7bab7bb3a1
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/80.9812/1
OpenGL version/vendor/renderer: 4.5.0 NVIDIA 353.30/NVIDIA Corporation/Quadro K2200/PCIe/SSE2
```