FreeCAD/FreeCAD #21121: CI: codespell isn't executing

luzpaz, 2025-05-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Doesn't look like codespell is being executed by the CI when triggered by a new PR or subsequent commits to a PR
Example: https://github.com/FreeCAD/FreeCAD/actions/runs/14764051621/job/41595309385?pr=21078#step:17:1

Returns: `Process completed with exit code 2.`

### Full version info

```shell
n/a
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-04:
@oursland @adrianinsaval FYI

oursland, 2025-05-04:
Looks like there were some missing command-line arguments in the implementation of `codespell.py`.  I have added them in PR https://github.com/FreeCAD/FreeCAD/pull/21127.

closed 2025-05-04 by chennes