FreeCAD/FreeCAD #21122: AddonManager: being blocked by Anubis from accessing macros on the wiki

luzpaz, 2025-05-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Reproduce:

* Open Addon Manager
* Search for honeycomb macro

Result:
[image]

ref: https://x.com/julsevern/status/1918779859955818944

### Full version info

```shell
n/a
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Papoow95, 2025-05-08:
Hello,
I'm also having trouble accessing the FreeCad.org forum server. I can't get past Anubis' control :(
Does anyone have a solution to this problem?

Thank you very much :)

luzpaz, 2025-05-08:
@maxwxyz should we have Anubis related ticket on this bugtracker or the homepage repo ?

fenio, 2025-05-10:
Hey folks.
First of all thanks for this great project. Amazing work and I really appreciate it.
Seems you're struggling with protecting your resources, forums, addons hosting and so on.
I'd like to help. I'm quite experienced sysadmin. Who's the best person to contact wrt these issues?
Who's responsible for FreeCAD infrastructure?

maxwxyz, 2025-05-10:
@chennes FYI

luzpaz, 2025-05-10:
@kkremitzki heads-up 👆

luzpaz, 2025-05-17:
The duplicate issues are racking up.

FlachyJoe, 2025-05-18:
Related Anubis doc: https://anubis.techaro.lol/docs/admin/policies/
A fix can be to define a specific useragent for the addon manager and white lists it on the server.

chennes, 2025-05-18:
However for that to work we almost certainly need https://github.com/FreeCAD/AddonManager/pull/72 -- otherwise we'll be allowing any robot that also uses the same user agent string as `QNetworkAccessManager` (which is very generic, and probably matches most robots out there).

whitlabell, 2025-06-03:
> Just to mention it's also affecting the python script for creating an offline copy of the Manual. The extract_manual_links() function is returning the links from the Anubis page rather than the manual homepage.
> 
> [https://wiki.freecad.org/FreeCAD_Manual_Converter](url)

Can be worked around by editing the two occurrences of "Mozilla" to "Curl" in the user agent string.

Reqrefusion, 2025-06-03:
Although I have a robot that performs a transaction on the wiki every 15 minutes, thankfully I did not encounter this problem. The working logic of Anubis is to tire the requestor, which makes requests costly and bots inefficient. Perhaps this can be done by changing some properties of the request.

Thovthe, 2025-07-21:
**This doesn't appear to be fully fixed.**  I've tried using the add-on manager to install the [Parametric Curve FP](https://github.com/mwganson/Parametric_Curve_FP/) macro which is on the [wiki](https://wiki.freecad.org/Macro_Parametric_Curve_FP) but it doesn't show up.

I've tried searching:
Parametric Curve FP
FP
Curv
Para
Parametric
Curve
Curves

Seems like an important macro, pity it doesn't work on the happy path.  Maybe it should be added to the github based macro repository.

### Freecad info:
OS: KDE Flatpak runtime
Architecture: x86_64
Version: 1.0.1.39285 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 878f0b8)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/

chennes, 2025-07-21:
If you use the Addon Manager to install a new copy of itself it should work for you.

Thovthe, 2025-08-03:
Thanks @chennes , the strange thing was that it seemed to get some of the wiki macros without installing the new version so I thought that the new addon manager maybe came with v1.0.1.  Having installed addon manager through itself indeed resolved the issue.