FreeCAD/FreeCAD #21136: Caps lock on does not allow me insert values to the numeric fields.

leoheck, 2025-05-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If capslock is on, I cannot type numeric values in numeric fields.

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.41464 (Git)
Build date: 2025/04/28 16:20:16
Build type: Release
Branch: main
Hash: 8d6547d519ffb331cfba054b4db2750b3e71339b
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.74.0
  * Curves 0.6.61
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * Curves.backup1746105636.3777354 0.6.60 (Disabled)
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.2.0
  * lattice2 1.0.0
  * kicadStepUpMod 11.6.9
```

### Subproject(s) affected?

Core

### Anything else?

This was reported previously here, but it happens on other parts of the software too.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-05-05:
I am unable to reproduce the issue. I tried with Draft_Rectangle and Part_Extrude tools.

```

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * BIM021 2021.12.0
  * Reporting
  * Telemetry 1.0.1

```

luzpaz, 2025-05-05:
> Version: 1.1.0dev.41365 (Git)

@semhustej can you try on a more recent macOS dev build ?

semhustej, 2025-05-05:
> > Version: 1.1.0dev.41365 (Git)
> 
> [@semhustej](https://github.com/semhustej) can you try on a more recent macOS dev build ?

Sorry I don't own Mac. I can only test on Windows :) . That's also the reason for the older weekly build - it is the latest Windows version.

kaiwas, 2025-05-05:
Manjaro Linux KDE.
I can't confirm. The numbers are entered normally

leoheck, 2025-05-06:
This also happens on Safe Mode.

When capslock is off, I can type numbers and letters in numeric fields.
Whe it is on I can only type uppercase letters

Not Safe mode.

https://github.com/user-attachments/assets/6c2f2e6a-a36d-4e1e-b1d8-d5678bd3bcdb

Safe Mode.

https://github.com/user-attachments/assets/2689b089-1016-47ce-894e-63cfa4796e0b

leoheck, 2025-05-06:
Saw there is a newer appimage, I am going to check that one.