FreeCAD/FreeCAD #21137: BIM: Arch_Roof edit mode uses mm independent of the unit system used

semhustej, 2025-05-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature**

**Problem description**
Arch_Roof edit mode uses mm independent of the unit system used. This is unexpected and brings inconvenience in transforming values for users. Users are not able to use different unit shortcut (such as cm, ft, etc.) in the edit dialogue mode, as is usual in different unit input fields in FreeCAD.

Steps to reproduce:

- select `Building Euro` as a unit system
- open new file
- create `Draft_Rectangle`
- create `Arch_Roof` from the rectangle
- enter the Roof edit mode
- all length units are in `mm`, even though the unit system is in `cm`. This is not desirable

[image]

Additionally:
- select one `Run` field
- type in `100 cm`
- FreeCAD shows error and it's not possible to edit the roof anymore
```
15:06:36  Traceback (most recent call last):
15:06:36    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41365-conda-Windows-x86_64-py311\Mod\BIM\ArchRoof.py", line 921, in edit
15:06:36      self.resetObject()
15:06:36    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41365-conda-Windows-x86_64-py311\Mod\BIM\ArchRoof.py", line 933, in resetObject
15:06:36      run.append(float(it.text(2)))
15:06:36                 ^^^^^^^^^^^^^^^^^
15:06:36  ValueError: could not convert string to float: '1m'
```

**Possible solution**
Roof edit mode should use units of the selected unit system.

Cherry on top: FreeCAD would not enter error when user types in something different than float in the roof edit fields.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * BIM021 2021.12.0
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-05:
Confirmed.
@semhustej thank you for always reporting well written issues!

semhustej, 2025-05-05:
> also fixed by [#21113](https://github.com/FreeCAD/FreeCAD/pull/21113) [@Roy-043](https://github.com/Roy-043) ?

I think this is not fixed by the PR - it is a different issue.

semhustej, 2025-05-31:
Hi,
the issue is fixed only from one half. 

- The units in the dialogue are shown in the current unit system units. This is correct.
- When user changes values and types in no unit, the unit is understood by FreeCAD as mm. In my opinion this makes the situation even more confusing than before.

See behavior in the video:

https://github.com/user-attachments/assets/89221b71-402e-4238-af7c-90b16d3e78a6

The desired behavior is that FreeCAD would automatically append the unit, which is set in the current unit system (as other Draft/BIM tools do).

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41866 (Git)
Build date: 2025/05/26 16:31:05
Build type: Release
Branch: main
Hash: bc5616dbf6cbe3a2d88992dc9572d6f5b5bb502f
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.1

```

Roy-043, 2025-06-02:
> The desired behavior is that FreeCAD would automatically append the unit, which is set in the current unit system (as other Draft/BIM tools do).

The situation for the other tools is somewhat different. Depending on how you activate the input boxes the content except the unit is selected. You can then just enter a numerical value and keep the unit. If however you enter a numerical value without a unit, the value is interpreted as mm as well.

I currently do not think it is possible to select only the numerical portion of a value in the widget used here. So if this is indeed deemed to confusing we should revert the PR.

semhustej, 2025-06-03:
Thanks for the explanation.

I am not convinced which one is better. I lean towards showing and using mm all the time (the original state - that would mean reverting the PR), as that is more consistent. 

- It takes less time to understand for a new user what is happening, because he/she can clearly see mm as a unit. 
- It would also mean no change for official release users. No change is better than change if benefits are doubtful

semhustej, 2025-06-04:
> Before the PR no units were indicated. IMO reverting is not a good idea. It may be possible to improve the PR but I have to do some more research, I do not know enough about Qt.

In previous version the units were set as a description of columns, so it was clear you are always working with mm:

[image]

As I stated I am not convinced which one is better, so let's keep the PR if you suggest so. The current version shows the units based on the current unit system correctly.

closed 2025-06-05 by chennes

semhustej, 2025-06-18:
Tested with latest weekly. Works great. Thank you!

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42180 (Git)
Build date: 2025/06/16 16:25:59
Build type: Release
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1

```

PR #21175 (2025-05-07): BIM: show units in Arch_Roof task panel
Fixes: #21137.

Note that the list values in the Property editor are unitless. Therefore lengths are still in mm there.
PR #21773 (2025-06-05): BIM: improve unit handling in Arch_Roof task panel
Fixes #21137.
Improves #21175.

Only the numerical portion of a cell's content is selected when the cell receives the focus. This is the same behavior we see in other Draft and BIM commands (Line, Wall, etc).

Additionally:
* The first and last column have been made read-only.
* Some columns have been widened.
* The text in the task panel has been reordered to match the order of the columns.
commit cfdb99fb (2025-06-05): BIM: improve unit handling in Arch_Roof task panel (#21773)

* BIM: improve unit handling in Arch_Roof task panel

Fixes #21137.
Improvement for #21175.

units are now kept when a cell receives the focus.

Additionally:
* The first and last column have been made read-only.
* Some columns have been widened.
* The text in the task panel has been reordered to match the order of the columns.

* _RoofTaskPanel_Delegate class only if Gui is up

* typo