FreeCAD/FreeCAD #21139: Sketcher in 1.1 weekly usability problem with copy/paste

theosib, 2025-05-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

See the sketch "server" in the attached design. I want to duplicate the E and R at the end of the word to spell SERVER. In the sketcher, I can select the existing E and R, copy, and paste. But when I paste, there are two major problems:

- The paste lands smack on top of the existing geometry, making it impossible to select just those parts using a rubber band.
- Imported geometry stays at the bottom of the list. So when I paste the 28 new objects, it's not intuitive as to which ones to select in the Elements panel.

Suggested ways to fix this:

- Be sure the newly pasted elements are selected in the Elements panel, allowing one to apply the move tool without having to figure out what to select.
- Possibly paste the duplicates where they don't overlap with existing geometry so that they can be selected by rubber band.

[server.FCStd.zip](https://github.com/user-attachments/files/20041305/server.FCStd.zip)

### Full version info

```shell
OS: macOS 15.3.2
Architecture: arm64
Version: 1.1.0dev.41464 (Git)
Build date: 2025/04/28 16:20:16
Build type: Release
Branch: main
Hash: 8d6547d519ffb331cfba054b4db2750b3e71339b
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: unset/Classic/macos
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * freecad.gears 1.3.0
  * Pyramids-and-Polyhedrons
  * fasteners 0.5.29
  * lattice2 1.0.0
  * Alternate_OpenSCAD 1.0.0
  * Curves 0.6.50
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-05-05:
I agree. The problem is very annoying.
One of the options is that the inserted objects will appear where the cursor is located. This would be very convenient.

Katt-Kushol, 2025-05-05:
Confirmed. Ctrl+C then Ctrl+V overlaps edges without any practical way to select the second set of edges.

> Suggested ways to fix this:
> 
> * Be sure the newly pasted elements are selected in the Elements panel, allowing one to apply the move tool without having to figure out what to select.

Maybe convert Ctrl+C command as the shortcut to the Move/Array transform (W) tool? That tool seems to do what are being suggested by the OP. 

[image]

kaiwas, 2025-05-05:
> Maybe convert Ctrl+C command as the shortcut to the Move/Array transform (W) tool? That tool seems to do what are being suggested by the OP.

No! This unfinished tool will break all your previously configured constraints.

The new tools are completely useless

https://github.com/user-attachments/assets/5562622b-e685-4097-8266-93febcb1c94a

theosib, 2025-05-05:
I really like the idea of the paste dropping the copy where the cursor is. I also think the new paste should have all new elements selected.

kaiwas, 2025-05-05:
I don't know at what point the insertion could be made. But most often it is the center of the boundy box. But what is also important is that after the insertion the object is selected.
Demonstration from inkscape.

https://github.com/user-attachments/assets/d050ef2d-7693-4e23-9625-587a725f28e9

theosib, 2025-05-05:
I would like credit someone named "bance" on Discord with an alternative approach to this. Instead of copying, what he did was select the part to copy and then use the move tool and set the copy count to 1. This left behind the original and allowed placement of a copy in another location.

https://github.com/user-attachments/assets/8c46618e-93f4-44e4-9da8-01ce34992117

PaddleStroke, 2025-05-06:
Copy paste has the advantage of working accross instances of freecad. So you can copy from one sketch into another freecad other sketch (you can even paste as text in the forum or such). Which the transform tool can't do.

I agree that the behavior should be as described : 
- Paste at mouse mosition.
- Pasted geometries are selected.

I would also add that it would be nice that : 
- a group drag is started after paste. I mean that when you paste, the geometries appear at mouse pointer but then they follow the mouse until the user clicks somewhere.

maxwxyz, 2025-06-20:
It definitely should stay an option to paste at the same location, e.g. in other sketches.