FreeCAD/FreeCAD #21140: The properties or taskbar cannot be minimized if its edge falls on the view tree.

kaiwas, 2025-05-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I encountered this problem when I reduced the window (made it floating).

If the tree view is located on the left, and the properties/tasks panel is located on the right - then after increasing the properties/tasks panel to the tree view borders, it cannot be reduced.

A workaround is to minimize and maximize the window.

Demonstration on the video.

https://github.com/user-attachments/assets/ee135ebe-3941-4862-be48-eebbd0da245f

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41464 (Git) AppImage
Build date: 2025/04/28 16:20:16
Build type: Release
Branch: main
Hash: 8d6547d519ffb331cfba054b4db2750b3e71339b
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * addFC
  * OpenTheme 2024.9.1
  * FreeCAD_SketchArch-master
  * PitchedRoof
  * Test
  * sheetmetal 0.7.22
  * Road 2025.2.22
  * Reinforcement
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-08:
@kaiwas could you check if https://github.com/FreeCAD/FreeCAD/issues/21141#issuecomment-2862017381 resolves it?

kaiwas, 2025-05-10:
> [@kaiwas](https://github.com/kaiwas) could you check if [#21141 (comment)](https://github.com/FreeCAD/FreeCAD/issues/21141#issuecomment-2862017381) resolves it?

The behavior does not depend on the settings.

https://github.com/user-attachments/assets/9509c65a-717e-49f3-8f45-0248df476cdd

Pendelf, 2025-08-29:
I confirm, the same behavior occurs on Windows 11 as well

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.43071 (Git)
Build date: 2025/08/25 18:08:30
Build type: Release
Branch: main
Hash: d85613b498881c47485a7d930807673203086756
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * addFC 1.9.1
  * CurvedShapes 1.0.13
  * Curves 0.6.70
  * fasteners 0.5.41
  * manifest.json
  * PieMenu 1.12.0
  * sheetmetal 0.7.24
  * StructureTools 0.0.1
  * SvgWorkbench 1.0.0.dev6
  * Telemetry 1.0.2

FEA-eng, 2026-09-16:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 Compatibility Profile Context 26.5.2.260413/ATI Technologies Inc./AMD Radeon(TM) Graphics
```