FreeCAD/FreeCAD #21141: Zooming with mouse wheel while above overlay doesn’t work

pierreporte, 2025-05-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Open a model or create a solid in a new document (doesn’t matter, it’s just to see something on screen). Then, zoom in or out with the mouse wheel (you may need to be in CAD navigation style). It works everywhere (the solid gets bigger on smaller) except on overlay zones, even if they are empty. Zooming using other way, even requiring only the mouse, works as intended.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41464 (Git) AppImage
Build date: 2025/04/28 16:20:16
Build type: Release
Branch: main
Hash: 8d6547d519ffb331cfba054b4db2750b3e71339b
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/117.596/1
Installed mods: 
  * Curves 0.6.59
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.21
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-06:
So the idea is, scroll if the overlay dialog is empty?
Or also, scroll if you hover over transparent parts in the overlay? This would lead to ending the scroll if you scroll past a shorter object name in the tree view or the task dialog if it has a break between sections.
@FreeCAD/design-working-group

MisterMakerNL, 2025-05-06:
I got dejavu, I am sure Realthunder fixed this in his linkstage branch.

pierreporte, 2025-05-07:
> So the idea is, scroll if the overlay dialog is empty? Or also, scroll if you hover over transparent parts in the overlay? This would lead to ending the scroll if you scroll past a shorter object name in the tree view or the task dialog if it has a break between sections. 

There could be a problem then. Since there are several navigation styles, it may not be a good idea to do that.

How about activate scrolling of the tree view only when the mouse is on a scrollbar at the left edge of the tree view (when overlay mode is active) or on the text?

kaiwas, 2025-05-08:
only now I realized that the problem I wrote about earlier after this topic is also related to the overlay (good timing)
https://github.com/FreeCAD/FreeCAD/issues/21140

hyarion, 2025-05-08:
> How about activate scrolling of the tree view only when the mouse is on a scrollbar at the left edge of the tree view (when overlay mode is active) or on the text?

We already have a setting for this in https://wiki.freecad.org/Preferences_Editor#UI "Auto mouse wheel pass through" and it works in opentheme.

Could you check if that settings is enabled?
Also do you have a background color for the active overlay? The auto-functionality should only swallow the event for opaque colors (which seems to be the default in freecad-dark theme?)

pierreporte, 2025-05-08:
@hyarion I see do difference when the option is checked or unchecked. The overlay is completely transparent.

maxwxyz, 2025-05-08:
so is there an issue?

PaxTux, 2025-10-17:
I think it's a regression.
In Preferences/Display/UI there is an option:
-In version 1.0.2 is called "Auto mouse wheel pass through" and works as intended.
-In version 1.1.0dev is called "Automatically pass through of the mouse wheel" and it has no effect. Mouse wheel event is not passed through the transparent part of the overlay panel, I tried with touchpad and actual mouse.

FEA-eng, 2025-11-23:
Another report: https://forum.freecad.org/viewtopic.php?t=101207