FreeCAD/FreeCAD #21149: Please add a reference to the clone tool on the scaling wiki page.

jgilmore, 2025-05-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Due to wiki write access being difficult to do, I'm reporting this here instead of making the change myself. 

The problem is that when searching for "How do I scale this damned part, it needs to be 120% larger!" the first hit that comes up is https://wiki.freecad.org/Part_Scale When the user actually needs to be pointed to https://wiki.freecad.org/Draft_Clone

Various forum topics about this issue include:
https://forum.freecad.org/viewtopic.php?style=3&t=59473
https://forum.freecad.org/viewtopic.php?t=81393
https://forum.freecad.org/viewtopic.php?style=3&t=3555

Meanwhile, the best reference I found (which turned me on to the clone's scaling parameters, which is what I was actually looking for) was https://www.xsim.info/articles/FreeCAD/en-US/HowTo/Scale-a-3D-object.html

No, it is not at all obvious that "scale" is done by and described in the "clone" wiki page, this needs to be made more obvious.asdfasdf

It would be help to change:
'''Part Scale''' scales shapes by a specified factor in all directions or by distinct factors in each cardinal direction. In the case of distinct factors, the shapes may be distorted.

to
'''Part Scale''' scales shapes by a specified factor in all directions or by distinct factors in each cardinal direction. In the case of distinct factors, the shapes may be distorted.

For some use cases, what you actually want is to clone the part, and use the clone's built-in "scale" parameter instead of this tool. See the "see also" links in the sidebar.

### Full version info

```shell
none, documentation issue.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-05-06:
> It would be help to change:
> '''Part Scale''' scales shapes by a specified factor in all directions or by distinct factors in each cardinal direction. In the case of distinct factors, the shapes may [...]

Both texts are the same AFAICT.

pitch-circle, 2025-05-07:
> Both texts are the same AFAICT.

It think the suggestion is to add the words below or something similar:

"For some use cases, what you actually want is to clone the part, and use the clone's built-in "scale" parameter instead of this tool. See the "see also" links in the sidebar."

So no editing of what is there, just addition of a short paragraph.

The suggestion could perhaps be simplified to say:

"Use of Draft, Clone, with its built-in scale parameter, instead of Part, Scale, may be preferred for some cases. [Draft, Clone to be a link]

Roy-043, 2025-05-07:
Thanks for clarifying.

I have edited the 1st paragraph. New text:

"Part Scale scales shapes by a specified factor in all directions or by distinct factors in each cardinal direction. In the case of distinct factors, the shapes may be distorted.

Scaling occurs relative to the global coordinate system. To scale relative to the placement of the source object create a Link or a Draft Clone instead."

closed 2025-06-17 by maxwxyz