FreeCAD/FreeCAD #21154: Draft: Cyclic build dependency BIM

pieterhijma, 2025-05-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

There are build dependencies that make it challenging to disable certain modules. Since FreeCAD takes quite some time to compile, it would be good to investigate the dependencies and make sure the defined dependencies are true dependencies.

The fact that there are cyclic dependencies may indicate that FreeCAD's modular design is not as modular as we think and that there are architectural design issues.

Example:

```
CMake Error at cMake/FreeCAD_Helpers/CheckInterModuleDependencies.cmake:11 (message):
  BUILD_DRAFT requires BUILD_BIM to be ON, but it is "OFF"
Call Stack (most recent call first):
  cMake/FreeCAD_Helpers/CheckInterModuleDependencies.cmake:22 (REQUIRES_MODS)
  CMakeLists.txt:74 (CheckInterModuleDependencies)
```

I consider Draft to be a more basic module than BIM. Therefore, I would expect BIM to depend on Draft, but not the other way around and certainly not both depending on each other.

From `CheckInterModuleDependencies.cmake`:

Cyclic dependency:

```
    REQUIRES_MODS(BUILD_BIM                BUILD_PART BUILD_MESH BUILD_MESH_PART BUILD_DRAFT)
    REQUIRES_MODS(BUILD_DRAFT              BUILD_SKETCHER BUILD_TECHDRAW BUILD_BIM)
```

I also find it suspicious that Draft depends on TechDraw but I can understand that a bit better because TechDraw is a C++ module, whereas Draft is not, so I can understand that Draft uses some API from TechDraw.

### Full version info

```shell
OS: Arch Linux (tty)
Architecture: x86_64
Version: 1.1.0dev.41585 (Git)
Build date: 2025/05/06 03:23:37
Build type: Unknown
Branch: main
Hash: 261ce705a93e98826389f20c4a99238757dfaab3
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 
shiboken , SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-05-07:
Indeed the Draft requiring BIM is an error.
Draft occasionally probes and uses BIM stuff here and there, but it is always checked at runtime and it always works without BIM.

I think we should simply remove the BIM dependency. Any error happening because of that should be treated as a bug

pieterhijma, 2025-05-08:
Ok, I'll create a PR.

chennes, 2025-05-14:
Does the Draft unit test suite still pass if BIM is disabled?

pieterhijma, 2025-05-14:
In the main branch the test suite has 1 error (with BIM enabled):

```
OS: Arch Linux (tty)
Architecture: x86_64
Version: 1.1.0dev.41697 (Git)
Build date: 2025/05/14 07:36:42
Build type: Unknown
Branch: main
Hash: 9d6f1ad37c1d97bcf669272c3124ce6045691b5b
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 6.9.0
shiboken 6.9.0, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
  * Curves 0.6.61
  * freecad.gears 1.3.0
```

```
20:08:08  ======================================================================
ERROR: test_copy (drafttests.test_modification.DraftModification.test_copy)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.13/unittest/case.py", line 58, in testPartExecutor
    yield
  File "/usr/lib/python3.13/unittest/case.py", line 651, in run
    self._callTestMethod(testMethod)
    ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/lib/python3.13/unittest/case.py", line 606, in _callTestMethod
    if method() is not None:
       ~~~~~~^^
  File "/home/pieter/scratch/build-freecad-main/Mod/Draft/drafttests/test_modification.py", line 79, in test_copy
    obj = Draft.move(line, c, copy=True)
  File "/home/pieter/scratch/build-freecad-main/Mod/Draft/draftfunctions/move.py", line 120, in move
    newobj = make_copy.make_copy(obj)
  File "/home/pieter/scratch/build-freecad-main/Mod/Draft/draftmake/make_copy.py", line 73, in make_copy
    newobj = App.ActiveDocument.copyObject(obj)
ValueError: {'sclassname': 'N4Base11PyExceptionE', 'sErrMsg': 'Empty module name', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'Empty module name', 'btranslatable': False, 'breported': False}
```

The branch of PR #21331 with BIM disabled I have 2 errors. The same one as above and a new one:

```
OS: Arch Linux (tty)
Architecture: x86_64
Version: 1.1.0dev.41697 (Git)
Build date: 2025/05/14 07:36:42
Build type: Unknown
Branch: fix-cyclic-build-dep
Hash: 9d6f1ad37c1d97bcf669272c3124ce6045691b5b
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 6.9.0
shiboken 6.9.0, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
  * Curves 0.6.61
  * freecad.gears 1.3.0
```

```
20:12:59  ======================================================================
ERROR: test_workingplaneproxy (drafttests.test_creation.DraftCreation.test_workingplaneproxy)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.13/unittest/case.py", line 58, in testPartExecutor
    yield
  File "/usr/lib/python3.13/unittest/case.py", line 651, in run
    self._callTestMethod(testMethod)
    ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/lib/python3.13/unittest/case.py", line 606, in _callTestMethod
    if method() is not None:
       ~~~~~~^^
  File "/home/pieter/scratch/build-freecad-fix-cyclic-build-dep/Mod/Draft/drafttests/test_creation.py", line 353, in test_workingplaneproxy
    obj = Draft.make_workingplaneproxy(placement)
  File "/home/pieter/scratch/build-freecad-fix-cyclic-build-dep/Mod/Draft/draftmake/make_wpproxy.py", line 52, in make_workingplaneproxy
    ViewProviderWorkingPlaneProxy(obj.ViewObject)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/home/pieter/scratch/build-freecad-fix-cyclic-build-dep/Mod/Draft/draftviewproviders/view_wpproxy.py", line 75, in __init__
    vobj.LineColor = params.get_param_arch("ColorHelpers") | 0x000000FF
                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
TypeError: unsupported operand type(s) for |: 'NoneType' and 'int'

20:12:59  
20:12:59  ------------------------------------------------------------------------------

20:12:59  ======================================================================
ERROR: test_copy (drafttests.test_modification.DraftModification.test_copy)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.13/unittest/case.py", line 58, in testPartExecutor
    yield
  File "/usr/lib/python3.13/unittest/case.py", line 651, in run
    self._callTestMethod(testMethod)
    ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/lib/python3.13/unittest/case.py", line 606, in _callTestMethod
    if method() is not None:
       ~~~~~~^^
  File "/home/pieter/scratch/build-freecad-fix-cyclic-build-dep/Mod/Draft/drafttests/test_modification.py", line 79, in test_copy
    obj = Draft.move(line, c, copy=True)
  File "/home/pieter/scratch/build-freecad-fix-cyclic-build-dep/Mod/Draft/draftfunctions/move.py", line 120, in move
    newobj = make_copy.make_copy(obj)
  File "/home/pieter/scratch/build-freecad-fix-cyclic-build-dep/Mod/Draft/draftmake/make_copy.py", line 73, in make_copy
    newobj = App.ActiveDocument.copyObject(obj)
ValueError: {'sclassname': 'N4Base11PyExceptionE', 'sErrMsg': 'Empty module name', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'Empty module name', 'btranslatable': False, 'breported': False}

20:12:59  
20:12:59  ------------------------------------------------------------------------------
```

furgo16, 2025-05-14:
Having added a few tests to BIM recently, I'm fairly certain that CI for Ubuntu native, and for pixi (IIRC Ubuntu and macOS) runs the full test suite, including Draft and BIM. I seem to recall it's skipped for Windows.

pieterhijma, 2025-05-15:
I agree with @yorikvanhavre in [his comment](https://github.com/FreeCAD/FreeCAD/issues/21297#issuecomment-2882887571). 

Since Draft requiring BIM is an error, I propose to merge https://github.com/FreeCAD/FreeCAD/pull/21331 and create issues for each of the problems that we already have encountered and are going to encounter as a result of this and fix those issues accordingly.

yorikvanhavre, 2025-05-15:
The failing test is probably because of a remaining `import Arch` somewhere

PR #21331 (2025-05-15): Fix a cyclic build dependency
This PR removes the cyclic dependency of Draft on BIM.

## Issues
Fixes #21154.

## Related information

After building with BUILD_BIM OFF, the following messages occur:

```
Preferences file :/ui/preferences-arch.ui not found
Preferences file :/ui/preferences-archdefaults.ui not found
Preferences file :/ui/preferences-dae.ui not found
Preferences file :/ui/preferences-ifc.ui not found
Preferences file :/ui/preferences-ifc-export.ui not found
Preferences file :/ui/preferences-sh3d-import.ui not found
```

See https://github.com/FreeCAD/FreeCAD/issues/21297.

I wou [... cut]
closed 2025-05-15 by hyarion