FreeCAD/FreeCAD #21158: PartDesign Hole Some checkboxes do not have right margin

leoheck, 2025-05-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The checkboxes "Include in depth" and "Switch Direction" do not have space between the text as the other checkboxes do.
Compare these 2 with the "Threaded" below.

See:

[image]

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.41464 (Git)
Build date: 2025/04/28 16:20:16
Build type: Release
Branch: main
Hash: 8d6547d519ffb331cfba054b4db2750b3e71339b
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.74.0
  * Curves 0.6.61
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * Curves.backup1746105636.3777354 0.6.60 (Disabled)
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.2.0
  * lattice2 1.0.0
  * kicadStepUpMod 11.6.9
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pierreporte, 2025-05-06:
It is because they are related to the drill angle entry. It has been done on purpose. Ping @alfrix

leoheck, 2025-05-06:
> It is because they are related to the drill angle entry. It has been done on purpose. Ping [@alfrix](https://github.com/alfrix)

Ah, the drill angle doesn't have the space either.

leoheck, 2025-05-06:
> Could you recheck with the built-in themes, not with a 3rd party theme?

Sure, as soon I get to the computer.

leoheck, 2025-05-06:
@maxwxyz in Safe Mode.

[image]

Tapered and Threaded look nice.
The ones checked do not look nice.

It might be related to a variable horizontal spacer or something that may be missing, right?

leoheck, 2025-05-06:
Also, the "Base profile types" is misaligned as well.

alfrix, 2025-05-06:
Those check boxes are custom (they can ellipsize), since the stylesheet reset all the values they need to be added alongside the selector for the checkboxes

leoheck, 2025-05-06:
Why "Include in depth" to elliosize? 
Also, I saw that other day, but today I was not able to redu

leoheck, 2025-05-06:
Sorry, closed the issue by mistake but I cannot reopen it.

alfrix, 2025-05-07:
> Why "Include in depth" has to elliosize?
> 
> Also, I saw that other day, but today I was not able to reduce this tasks panel less than that, so those items are not ellipsizing since the min width of the panel seems fixed.

translations... english can be short but other languages usually are much longer, the minimum size of the panel will depend on whats visible

PR #21230 (2025-05-16): Gui: fix ElideCheckbox
This fixes #21158 and also some special states that weren't draw like hover and focus
closed 2025-05-16 by kadet1090