FreeCAD/FreeCAD #21160: Restart back from Safe mode.

leoheck, 2025-05-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using "Normal" mode, I can restart Freecad in Safe mode.
However, when in Safe mode, I cannot restart back to Normal mode.

Can we have such an option?

Actually, in Safe Mode, it should not show the `Restart in Safe mode`.
This option should be suppressed, I guess, or replaced with this alternative I would like to have.

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.41464 (Git)
Build date: 2025/04/28 16:20:16
Build type: Release
Branch: main
Hash: 8d6547d519ffb331cfba054b4db2750b3e71339b
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-05-06:
I think it's reasonable to expect user to restart in safe mode twice. As an example:

- I `Restart in safe mode` and change settings and check how FreeCAD works
- now I want to see how FreeCAD works with default settings, or check different settings than currently set
- I use `Restart in safe mode` again

leoheck, 2025-05-06:
Hm, makes sense. This doesn't invalidate having an option to go back to the normal mode.

luzpaz, 2025-05-06:
* https://github.com/FreeCAD/FreeCAD/issues/6106#issuecomment-2423342525

closed 2025-05-06 by maxwxyz

luzpaz, 2025-05-07:
@maxwxyz I don't see this mentioned in #6106. If this is closed - and it's deemed a valid feature request -  then it should be added to #6106, right ?

maxwxyz, 2025-05-07:
@luzpaz it is right in the first image:

[image]

leoheck, 2025-05-07:
Yes right. But you showed something with something else. And then I gave my opinion as well on this picture, since the "name of the action" doesn't look good if it is related to the topic of course.