FreeCAD/FreeCAD #21167: Using Cmd + Q to quit FreeCAD causes a freeze

hsvfan-jan, 2025-05-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

On macOS 15.4.1, using `Cmd` + `Q` to quit the program, freezes it. Clicking `Quit freecad` from the menu does not have the same problem.

How to reproduce: just open the program and close it, no need to open a file or anything.

### Full version info

```shell
Development Build weekly-2025.05.06
```

### Subproject(s) affected?

None

### Anything else?

I saw https://github.com/FreeCAD/FreeCAD/issues/21094 but it seems to be a different bug.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-06:
Is this a dupe, @maxwxyz  ?

maxwxyz, 2025-05-06:
I don't think so.
@hsvfan-jan please provide full about info.

hsvfan-jan, 2025-05-06:
@maxwxyz what other info do you need?
It should be easily reproducible in a few seconds without any special setup.

Open FreeCAD --> press `Cmd + Q` --> freeze happens

JohnOCFII, 2025-05-06:
I'm not seeing the issue.  I can quit properly with either `CMD-Q` or selecting `File --> Quit`

```
OS: macOS 15.4.1
Architecture: arm64
Version: 1.1.0dev.41585 (Git)
Build date: 2025/05/06 03:23:37
Build type: Release
Branch: main
Hash: 261ce705a93e98826389f20c4a99238757dfaab3
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2024.9.1
  * AddonManager 2025.5.29a
  * Telemetry 1.0.1
```

hsvfan-jan, 2025-05-06:
Hm, earlier I reproduced it three times even *after* finding https://github.com/FreeCAD/FreeCAD/issues/21094 so I thought I made sure not to hover over anything that'd cause a pop-up and hence a freeze.

Copying the about section causes 5-10 second freeze but it recovered from it (should an issue be opened for this? seems awfully long). Anyway, here's the info.
```
OS: macOS 15.4.1
Architecture: arm64
Version: 1.1.0dev.41585 (Git)
Build date: 2025/05/06 03:23:37
Build type: Release
Branch: main
Hash: 261ce705a93e98826389f20c4a99238757dfaab3
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/Germany (en_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
```
If I can't reproduce it in the following days, I'll close this issue as duplicate for #21094.

JohnOCFII, 2025-05-06:
> Copying the about section causes 5-10 second freeze but it recovered from it (should an issue be opened for this? seems awfully long).

Already open - in case you wanted to add your experience: https://github.com/FreeCAD/FreeCAD/issues/20837

hsvfan-jan, 2025-05-13:
Can't replicate anymore, seems to be fixed at the latest by 2025.05.1
```
OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.41671 (Git)
Build date: 2025/05/12 16:21:14
Build type: Release
Branch: main
Hash: a997c1ef16a0a4657e0edf61fb15d01058832227
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Germany (en_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
```

closed 2025-05-13 by hsvfan-jan