FreeCAD/FreeCAD #21172: Units: Schema index issue regression possibly due to PR20540

Syres916, 2025-05-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Running the following Python code in the Console:

`App.Units.schemaTranslate(App.Units.Quantity("1 mm"), App.Units.getSchema())`

should produce the following output:

`('1.00 mm', 1.0, 'mm')`

not:

`invalid schema index:10`

Based on the last good build:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41464 (Git) AppImage
Build date: 2025/04/28 16:20:16
Build type: Release
Branch: main
Hash: 8d6547d519ffb331cfba054b4db2750b3e71339b
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.1131/1
Installed mods: 
  * FreeCAD-Ribbon 1.8.1.2 (Disabled)
  * btl 0.9.9
  * FreeCAD-themes 2025.1.7
  * Silk 0.1.6
  * sheetmetal 0.7.22
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.30.3
  * ChangeAppFont
  * Quetzal 1.4.4
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * Curves 0.6.61
  * QetWireManager
  * Assembly4 0.50.18
  * A2plus 0.4.68
  * ThreadProfile 1.94.0
```

it's a very recent regression, https://github.com/FreeCAD/FreeCAD/commit/1155f0d75281e9b9d70536d9b81dde7caececadd would be logical commit to investigate.

### Full version info

```shell
Hash: 1155f0d75281e9b9d70536d9b81dde7caececadd
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-05-06:
Unit Test proposal https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Simple_Unit_Schema_Test

CC @3x380V

Syres916, 2025-05-06:
Having reverted the above Hash and recompiled, I can confirm that PR20540 is the cause of the regression.

3x380V, 2025-05-06:
Indeed (they are shooting on the messenger already :))
```
>>> App.Units.schemaTranslate(App.Units.Quantity("1 mm"), 9)
('1.00 mm', 1.0, 'mm')
```
I will look at this in the morning.

sgahf, 2025-05-07:
Probably a related issue : having trouble to place a LCS on a face, for example using XY or Inertial placement, and trying to offset this LCS with Fx values.... The GUI reject the value telling me there is a problem with the unit. Knowing the fact that my Fx include some "/" in calculation...

3x380V, 2025-05-07:
@sgahf, could you try the linked fix? And in case the problem persist, open issue for it and provide test case. Thank you.

sgahf, 2025-05-07:
I don't have dev environment for that...
Can you package an AppImage for tests ?

sgahf, 2025-05-07:
Alternatively : here is a file which have the problem.
If you can add some parts or change some links in the assembly, should be okay...

[wagon.tombereau.zip](https://github.com/user-attachments/files/20087388/wagon.tombereau.zip)

closed 2025-05-14 by hyarion

hyarion, 2025-05-15:
> Just a short message to confirm problem is now solved un daily release 2025.05.15. Thank you for your efficient help !

thanks for confirming, it's really appreciated!

PR #21174 (2025-05-14): Base: fix Unit scheme management regressions
Fixes: #21172
Fixes: #21209
Fixes: #21245
Fixes: #21316
Fixes: #21317