FreeCAD/FreeCAD #21173: PartDesign: Fillet and Subtractive Pipe are failing to make a fillet.

leoheck, 2025-05-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have this simple but irregular shape that I am failing to add a fillet.
File: [test_Body_top_fillet.FCStd.zip](https://github.com/user-attachments/files/20070769/test_Body_top_fillet.FCStd.zip)

Side view

[image]

Front view

[image]

I want to make a fillet around this highlight edge..

[image]

The fillet tool is not working, no matter the radius of the fillet.

The Subtractive Pipe tool with something like this is also failing.

[image]

[image]

It is doing something, but not the right thing, see.

[image]

I don't even have the "Allow multiple solidis" enabled in the settings.

[image]

To make it simpler, I am not even using a complicated path for the pipe tool
I am using this flat square with rounded corners.

[image]

Is it a bug?
Is there any workaround that I can use to make this filet?

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.41573 (Git)
Build date: 2025/05/05 17:45:01
Build type: Release
Branch: main
Hash: 3189030c11010631e726dbe350e9077dc84109bb
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.74.0
  * Curves 0.6.61
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * Curves.backup1746105636.3777354 0.6.60 (Disabled)
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.2.0
  * lattice2 1.0.0
  * kicadStepUpMod 11.6.9
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-05-06:
I also tried this by selecting the edges to create a sub-shape binder.
The pipe tool proposes something that I would like to have.
But the resulting object is not ok.

[image]

[image]

leoheck, 2025-05-06:
The Additive Pipe fails to add mass in a different way.

[image]

leoheck, 2025-05-06:
Ah, changing the profile I could see the issue. The path is not closed. Is it right?

[image]

leoheck, 2025-05-06:
Ah, it works if I remove a section of the path.
[image]

The bottom_left corner
[image]

Ok, I can make 2 paths to make this work.

Now I understand what it means when it says profile is overlapping. To close the path, it has to overlap a bit...
This would not be a problem if it could work like the Pattern Tools I guess.

leoheck, 2025-05-06:
Awesome, seems good enough.

[image]

leoheck, 2025-05-06:
Ok, so question. Is this a bug, or should I close this.
I would like to see this working with a single path, if possible.

leoheck, 2025-05-07:
Why I get these things? Unless this window reports the same things as the Report View.

leoheck, 2025-05-07:
The Sketch492, for instance, is a simple, fully constrained sketch that does not give me any issue when analyzed.
Which version are you using @maxwxyz ?

leoheck, 2025-05-07:
When I bring this solution to my project. It does not behave like when it was in a reduced form on a different file.
Features look like they will be ok, but the mass is not removed.

[image]

Features started to remain active together without being edited.
FreeCAD has been open since yesterday, it may be related, or not.

[image]

leoheck, 2025-05-07:
I don't even have a lack of resources. 
[image]

maxwxyz, 2026-01-08:
@pieterhijma is this an OCC issue?

stefano2734, 2026-09-17:
I see here only last test with 2 years old occt 7.8.1.
https://github.com/Open-Cascade-SAS/OCCT/releases/tag/V7_8_1
Occt 7.9.3 and 8.0.1 are here actual available with some solved issues also to FreeCad. 
They are available with FreeCad Libpack 3.4.0 and 3.5.5.
Improvement to fillets will come with PRs in 8.0.2 and 8.1.0.