FreeCAD/FreeCAD #21179: BIM: Object library doen't closing even aftrer document closed

ErtugrulEge, 2025-05-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings community!
Even though I close the document, the objects library does not close

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * Curves 0.6.61
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-05-07:
I think this might be related. The PR was merged 2 days ago, so it is not yet in the weekly.
https://github.com/FreeCAD/FreeCAD/issues/17952

maxwxyz, 2025-05-08:
Please check in next weeks dev version

Roy-043, 2025-05-08:
I am afraid #17952 does not fix this issue. The BIM_Library command has a different logic. The same no doubt applies to other BIM features and commands. Finding a solution for more commands may be possible, I have to check.

closed 2025-05-12 by yorikvanhavre

semhustej, 2025-05-17:
I did a test and the issue is fixed. Thank you.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41694 (Git)
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.1

```

PR #21253 (2025-05-12): BIM: close BIM_Library task panel on doc close
Fixes #21179.

This PR uses the built-in solution to close the task panel.

I was not aware of this feature before. My 2 previous PRs related to the closing of task panels, #20719 and #21073, will (largely) have to be reverted/redone in favor of this solution.
commit ca0630aa (2025-05-12): BIM: close BIM_Library task panel on doc close

Fixes #21179.

This PR uses the built-in solution to close the task panel.

I was not aware of this feature before. My 2 previous PRs related to the closing of task panels, #20719 and #21073, will (largely) have to be reverted/redone in favor of this solution.