FreeCAD/FreeCAD #21180: BIM: Object library: Inserting an object gives error messages

ErtugrulEge, 2025-05-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings community!

Object library objects seems working but gives error messages

To reproduce :
1.Create an empty BIM file
2. Open Objects Library
3. I selected Architectural parts->Bedroom->Bedrom_closet (or you can try anything)

gives too many opened non-intrusive notifications 

The reprot view looks like this:

```
09:52:10  <ElementMap> BimLibrary.py(589): Invalid element name string id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Invalid element name string id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Invalid element name string id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Invalid element name string id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Invalid element name string id
09:52:10  Reading failed from embedded file: Compound.Shape.Map.txt (3604 bytes, 915 bytes compressed)
09:52:10  Reading failed from embedded file: Compound001.Shape.Map.txt (5203 bytes, 1296 bytes compressed)
09:52:10  <ElementMap> BimLibrary.py(589): Invalid element name string id
09:52:10  <ElementMap> BimLibrary.py(589): Invalid element name string id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Invalid element name string id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Invalid element name string id
09:52:10  Reading failed from embedded file: Compound004.Shape.Map.txt (2913 bytes, 848 bytes compressed)
09:52:10  Reading failed from embedded file: PathArray041.Shape.Map.txt (3267 bytes, 935 bytes compressed)
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Invalid element name string id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  Reading failed from embedded file: Compound005.Shape.Map.txt (3535 bytes, 800 bytes compressed)
09:52:10  Reading failed from embedded file: PathArray043.Shape.Map.txt (2694 bytes, 764 bytes compressed)
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  Reading failed from embedded file: Compound006.Shape.Map.txt (3969 bytes, 1008 bytes compressed)
09:52:10  Reading failed from embedded file: PathArray045.Shape.Map.txt (4330 bytes, 1094 bytes compressed)
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
09:52:10  <ElementMap> BimLibrary.py(589): Missing element name prefix id
[... 160 lines cut ...]

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * Curves 0.6.61
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-05-07:
This is due to the version of FreeCAD that the file was fully Recompute'd and then saved (1.0). We can't update the Library objects to 1.1dev _now_ otherwise 1.0.x users will get a sea of red due to ViewProviders that don't exist in the Stable version.

Syres916, 2025-05-07:
A compromise might be up update all the objects when the packagers decide on the release date of 1.0.1.

maxwxyz, 2025-05-08:
@yorikvanhavre FYI

Roy-043, 2025-08-30:
AFAICT the warnings fall into 3 categories:
1. ElementMap warnings: These are TNP related. They do not occur if you open the library document, they occur if you merge the library document into another document. Saving the library document in the current dev version does not solve them.
2. v1.1 etc. warnings: These are intended to alert the user that a Draft or BIM object has new or modified properties in the current version. It was never considered that there could be this many. I'll suppress them by turning them into log messages. Saving the library document in the current dev version avoids them as well obviously.
3. "End Offset too large for path length minus Start Offset. Using 0 instead." warnings. These do not occur if you save the library document in the current version. I have yet to find out why that is.

```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

closed 2026-09-04 by maxwxyz