FreeCAD/FreeCAD #21181: Attachment on geometric elements is not possible anymore

maxwxyz, 2025-05-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

It is not possible anymore to attach objects (e.g. datum plane or LCS) on selected geometry elements (e.g. faces or edges) in the document.
The mode is always "Deactivated".

[image]

Also, old files are broken when opened in the current weekly:

```
10:12:26  PositionBySupport: AttachEngine3D: subshape not found Pad001.Face7
10:12:26  <Exception> Attacher.cpp(890): AttachEngine3D: subshape not found Pad001.Face7
10:12:26  DatumPlane: AttachEngine3D: subshape not found Pad001.Face7

```

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.41585 (Git)
Build date: 2025/05/06 03:23:37
Build type: Release
Branch: main
Hash: 261ce705a93e98826389f20c4a99238757dfaab3
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Cables 0.1.4
  * CfdOF 1.30.1
  * Curves 0.6.61
  * Defeaturing 1.2.4
  * OpticsWorkbench 1.2.0
```

### Subproject(s) affected?

Core

### Anything else?

@PaddleStroke FYI

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-05-07:
See https://github.com/FreeCAD/FreeCAD/issues/21182#issuecomment-2857981873

maxwxyz, 2025-05-07:
also confirmed on
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41585 (Git)
Build date: 2025/05/06 03:23:37
Build type: Release
Branch: main
Hash: 261ce705a93e98826389f20c4a99238757dfaab3
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/133.772/2
Installed mods: 
  * Cables 0.1.4
  * CfdOF 1.30.1
  * CurvedShapes 1.0.13
  * Curves 0.6.61
  * dodo 1.0.1
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * Gridfinity 0.11.7
  * Manipulator 1.5.9
  * OpticsWorkbench 1.2.0
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.22
  * Silk 0.1.5
  * Telemetry 1.0.0
```

luzpaz, 2025-05-15:
This is windows specific, right ?

maxwxyz, 2025-05-15:
I assume, yes

maxwxyz, 2025-06-30:
this is with the release on the main repo, can you check on that?

sclebo05, 2025-06-30:
I've been following this bug as I have model that is affected. I noticed last week that the Freecad-Bundle Weekly Build 42325 no longer had the issue, but the official weekly in this repo (2025.06.23) still had the issue. In fact I can't create a simple cube on X_Y with build 2025.06.23.  

Looked at the git log between the two, short list and didn't see a reason for the differing behavior. I did not check libraries, etc as my Help->About->Libraries is blank on Windows.

$ git log 6faa377..f3980c8 --oneline
f3980c8a6f Merge pull request #21732 from tarman3/engrave
27c786fef1 Draft : fix Label Justification (#13444)
c55102acf8 Merge branch 'main' into engrave
96b006a2dd Spreadsheet Fix '=' regression in spreadsheets (#22146)
7545c9fd23 Merge pull request #21961 from knipknap/tool-cleanups
580d538798 Sketcher: Add contextual input hints to constraint commands (InputHints Phase 2) (#21751)
84f32bfcd8 Refactor slot op. (#21799)
a1636cbc27 Merge pull request #21943 from tetektoza/fix/21927_change_enter_ovp_accepting_dim
2069e799bb (tag: weekly-2025.06.23) CAM: Load preferences before activating the workbench (#21981)
75017b2dd2 Sketcher: Make TAB clear the field if user hasn't valid input
4fcb871b5b Spreadsheet: Only use validator when when prefix is not '='
a39c44e483 Tools: Add ExpressionLineEdit to QtDesigner plugin
7515c70b67 Sketcher: Make TAB lock the label if user has typed previously
9912c029dc Sketcher: Add QLabel header for newly added locked icon
78a754e620 Sketcher: Force cycling back to first labels on OVP if they are not set
1f250fab32 Sketcher: Initialize lock icon only once
bcbf8d88a9 Sketcher: Adjust P&D mode to new enter behavior
ef0b259097 Sketcher: Change enter behavior on OVP to put OVP in lock state only
a56357e0ff CAM: Engrave - Fix step down for closed profile
b88bafc138 CAM: fix: broken import in tests
c7a27e8b82 CAM: Remove obsolete images
07d466e1fb CAM: move DetachedDocumentObject into a dedicated module for better reuse

Just confirmed that the newest Freecad-Bundle Weekly Build 42340 is also 'fixed' for Windows. I'm not a dev, just a casual user, so ignore me if this isn't helpful!

maxwxyz, 2025-07-01:
@oursland did something change in the main builds?

sclebo05, 2025-07-08:
Tested the 2025.07.07 build, and this issue remains resolved. Fixed?