FreeCAD/FreeCAD #21182: Transform: Align dragger to custom element not possible anymore

maxwxyz, 2025-05-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

It is not possible anymore to align the dragger in the Std_Transform dialog to any custom geometry element in the document. On select and click nothing happens, there is no error or warning.
Maybe related to #21181 @kadet1090 ?

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.41585 (Git)
Build date: 2025/05/06 03:23:37
Build type: Release
Branch: main
Hash: 261ce705a93e98826389f20c4a99238757dfaab3
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Cables 0.1.4
  * CfdOF 1.30.1
  * Curves 0.6.61
  * Defeaturing 1.2.4
  * OpticsWorkbench 1.2.0
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-05-07:
Works fine using:

```
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41594 (Git)
Build date: 2025/05/06 15:50:21
Build type: Release
Branch: main
Hash: 998f4e4d451b850ab14dd9e34de8c365af406cb3
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_87, Eigen3 3.4.0, PySide 
shiboken , SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.9222/1
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * sheetmetal 0.7.22
  * Reinforcement
  * fasteners 0.5.38
```

Looks like another Windows Toponaming packaging issue to me, the same for #21181 which also works as expected using the above build.

maxwxyz, 2025-06-30:
this is with the release on the main repo, can you check on that?

FEA-eng, 2025-06-30:
> this is with the release on the main repo, can you check on that?

That could explain it, I use the FreeCAD Bundle weeklies because the Windows ones in the main repo are severely broken (can't even pad anything).

Regarding the issue, I can't check it in those main repo builds yet because the Transform tool is still not working there ("No extension of given type available" error already fixed in FreeCAD Bundle). I will try tomorrow.

maxwxyz, 2025-06-30:
so if we close it we say there are no issues with the builds on main (where we want them and they should work). It's like saying: close the issue if you cannot build FreeCAD on your OS, as you can download it elsewhere.

FEA-eng, 2025-06-30:
> so if we close it we say there are no issues with the builds on main (where we want them and they should work). It's like saying: close the issue if you cannot build FreeCAD on your OS, as you can download it elsewhere.

I also think it shouldn't be closed until it's fixed in the main repo builds.

sclebo05, 2025-06-30:
Definitely want the main repo builds to be functional. Do comparisons to Freecad-Bundle help identify a difference in libraries or build?

maxwxyz, 2025-07-08:
Also works in the windows builds on main now.
I think they are functional now @oursland