FreeCAD/FreeCAD #21185: FEM: mesh export units

adrougk, 2025-05-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have created a 3D geometry using [m] as the unit of measurement. I have generated a FE mesh, using both Gmsh and Netgen. When exporting the mesh, the node coordinates appear to be in [mm]. Changing the project system of units does not appear to affect the exported file coordinates.

Ideally, the exported coordinate units should match the ones used in the project in order to avoid having to convert them later.

Thanks!

[example.zip](https://github.com/user-attachments/files/20080812/example.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-08:
@FEA-eng FYI

FEA-eng, 2025-05-12:
What format are you using to export the mesh? That's a limitation rather than a bug - FreeCAD internally uses SI(mm) units, particularly for FEM (see how e.g. CalculiX's input deck is written). A preference to export in another unit system could be useful though.

adrougk, 2025-05-12:
I have tried .meshpy, .txt (Z88), .dat, all with the same result: node coordinates are in [mm] regardless of the project unit system.