FreeCAD/FreeCAD #21187: BIM: Please advise why user editable parameters such as Profile - Width have been locked, this is causing users to not be able to use legitimate Gui interation such as changing Profile type to work as they should

Syres916, 2025-05-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

Start a new Project File, switch to the BIM Wb and click the Profile icon
At X0, Y0, Z0 use Category INP, Preset INP 80

Double-click the INP80_ object in tree and change the Preset to INP200 which works fine in 1.0 but fails using current main with error:

```
12:37:14  pyException: Traceback (most recent call last):
  File "/home/username/freecad-daily-build/Mod/BIM/ArchProfile.py", line 446, in accept
    _ProfileH(self.obj, self.Profile)
  File "/home/username/freecad-daily-build/Mod/BIM/ArchProfile.py", line 154, in __init__
    obj.addProperty("App::PropertyLength","Width","Draft",QT_TRANSLATE_NOOP("App::Property","Width of the beam"), locked=True).Width = profile[4]
<class 'NameError'>: {'sclassname': 'N4Base9NameErrorE', 'sErrMsg': 'Property Unnamed#Profile.Width already exists', 'sfile': '/home/username/freecad-daily-source/src/App/DynamicProperty.cpp', 'iline': 209, 'sfunction': 'App::Property* App::DynamicProperty::addDynamicProperty(App::PropertyContainer&, const char*, const char*, const char*, const char*, short int, bool, bool)', 'swhat': 'Property Unnamed#Profile.Width already exists', 'btranslatable': False, 'breported': False}
```

There are multiple BIM commands that this `locked=true` has been applied to, I'm struggling to believe that this was intended.

### Full version info

```shell
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41594 (Git)
Build date: 2025/05/06 15:50:21
Build type: Release
Branch: main
Hash: 998f4e4d451b850ab14dd9e34de8c365af406cb3
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_87, Eigen3 3.4.0, PySide 
shiboken , SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.9222/1
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * sheetmetal 0.7.22
  * Reinforcement
  * fasteners 0.5.38
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-05-07:
@semhustej if you're aware of any other problems with the settings locked by https://github.com/FreeCAD/FreeCAD/pull/20778/commits/4a61a49456501a4434c14ac94e5dd6870a554dc7 then by all means feed them back I'm sure Profile isn't the only one.

semhustej, 2025-05-07:
I ran into this two weeks ago: https://github.com/FreeCAD/FreeCAD/issues/21001

I don't know of any other problematic place, as don't really understand the commit in-depth.

FlachyJoe, 2025-05-07:
Hi,
 `lock=True` set the _DynamickLock_ status for the property. With this status set the end-user can't **RMB/Remove property** on it (if the user did so, they broke the object).

But it also lock python `removeProperty` which return `False`.
It seems the BIM workbench doesn't test the `removeProperty` return value and assumes it can add a new one with an already used name.

To fix this we could:

-  don't add an already existent property but change its value

or 

-  unset the _DynamickLock_ status before removing the property

Syres916, 2025-05-07:
Thanks for the feedback @FlachyJoe , I'll have a look at the more complex Draft and CAM task panel updates tomorrow, I think we need a consistent method across all core Python modules.

PR #22052 (2025-06-23): BIM: fix ArchProfile update issues
Fixes #21001
Fixes #21187
closed 2025-06-23 by yorikvanhavre