FreeCAD/FreeCAD #21191: PartDesign: Back hole for Blind holes

leoheck, 2025-05-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

PD Hole is getting better and better. This feature could be incorporated as well.

I want to propose having settings to make the back of blind holes open like in this example below.
It could work like the Counterbore Head (except that the Counterbore head still does not have a taper angle for injection moulding)

It could have a length of separation or thickness below the Hole, and also a diameter, its length or depth, and a taper angle.

[image]

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.41573 (Git)
Build date: 2025/05/05 17:45:01
Build type: Release
Branch: main
Hash: 3189030c11010631e726dbe350e9077dc84109bb
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.74.0
  * Curves 0.6.61
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * Curves.backup1746105636.3777354 0.6.60 (Disabled)
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.2.0
  * lattice2 1.0.0
  * kicadStepUpMod 11.6.9
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pierreporte, 2025-05-07:
Your idea is similar to what Solidworks calls [advanced holes](https://help.solidworks.com/2024/English/SolidWorks/sldworks/c_advanced_holes_divider.htm). It would indeed be interesting to have the equivalent in FreeCAD.

pierreporte, 2025-05-07:
It could be a good idea to make the issue more generic

leoheck, 2025-05-07:
Oh, nice feedback. 
@pierreporte, how can I make this issue more generic?

pierreporte, 2025-05-07:
You can get inspiration from the Solidworks documentation linked in my comment. But first I would make it more generic by specifying that the holes can be through, and that you can have a different head on each side, and a thread on each side (different on each side). At the end of the day it’s up to you.

Ping @FreeCAD/cad-working-group

leoheck, 2025-09-04:
Ping @FreeCAD/cad-working-group.