FreeCAD/FreeCAD #21196: PartDesign: FC complains when using DatumLine for Pull Direction

leoheck, 2025-05-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The pull direction is a datum line, but it complains.

[image]

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.41573 (Git)
Build date: 2025/05/05 17:45:01
Build type: Release
Branch: main
Hash: 3189030c11010631e726dbe350e9077dc84109bb
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.74.0
  * Curves 0.6.61
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.2.0
  * lattice2 1.0.0
  * kicadStepUpMod 11.6.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-05-07:
Sketch works, another feature of the design, works as well.

leoheck, 2025-05-07:
I will check this later with a smaller file, cool?

leoheck, 2025-05-08:
The same still applies, unless I am doing something wrong.
[lala.FCStd.zip](https://github.com/user-attachments/files/20103892/lala.FCStd.zip)

luzpaz, 2025-05-17:
I need steps to recreate. I only manage to see:  
`08:52:05  Empty draft created !`

leoheck, 2025-06-11:
This is still happening.

Datum line was selected as Pull Direction.

Also, I cannot select and use `Z-Axis` directly instead. WHY?

Also, the name Datumline means what? 
It should have at least a label there, since any dadum line will appear as `Datumline`

[image]

```
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.42006 (Git)
Build date: 2025/06/02 16:32:08
Build type: Release
Branch: main
Hash: db1b5a877314f8d21514680421649fd9dc65acde
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
```

leoheck, 2025-06-11:
> I need steps to recreate. I only manage to see:

@luzpaz come on dude , a file was shared above, just open it and see the issue.

But here are the steps.
1. Create a Body
2. Create a Sketch with a Shape lets say a circle.
3. Pad it.
4. Select the vertical face (the side of the created cylinder)
5. Use the PartDesign Draft tool
6. Select the neutral plane (the top or bottom of the cylinder)
7. For the pull direction, select Z-Axis (no you can't)
8. Create a DadumLine using the Z-Axis as reference (rename it to My_Datum_Line)
9. Edit the Draft feature again.
10. For the pull direction, use the My_Datum_Line
11. Check the Report View.