FreeCAD/FreeCAD #21202: About: Pyside and Shiboken are not being reported in the About info

luzpaz, 2025-05-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

```
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
```

### Full version info

```shell
OS: Manjaro Linux (KDE/plasmawayland/wayland)
Architecture: x86_64
Version: 1.1.0dev.41573 (Git) AppImage
Build date: 2025/05/05 17:45:01
Build type: Release
Branch: main
Hash: 3189030c11010631e726dbe350e9077dc84109bb
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.065/1
Installed mods: 
  * fcVM
  * FreeCAD-Ribbon 1.3.17.1
  * SearchBar 1.0.2
  * fcVM.backup1738519851.1523638 (Disabled)
  * Silk 0.1.5
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-08:
Unrelated but would be nice to have  
* #18856

Syres916, 2025-05-08:
@luzpaz ideally the title should specify _Qt6 builds_ due to the code only looking for PySide2 and Shiboken2 versions see https://github.com/FreeCAD/FreeCAD/blob/main/src/LibraryVersions.h.cmake#L43 and https://github.com/FreeCAD/FreeCAD/blob/main/src/LibraryVersions.h.cmake#L49

luzpaz, 2025-05-08:
@Syres916 thanks for the heads up.

Syres916, 2025-05-08:
I can only wait for the CI Tests to complete but I think it's simply removing the `2` from each line as per https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Gui_About_PySide_Shiboken_All_Versions

Syres916, 2025-05-08:
New output on Linux but need Windows and Mac CI Tests to complete to ensure there's nothing specific regarding those platforms :

```
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41605 (Git)
Build date: 2025/05/07 15:23:29
Build type: Release
Branch: main
Hash: ffd53ee75f697189181a90d9f43487e3a36452f9
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 6.9.0
shiboken 6.9.0, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.9222/1
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * sheetmetal 0.7.22
  * Reinforcement
  * fasteners 0.5.38
```

luzpaz, 2025-05-08:
@Syres916 so we just need to have supportive logic for Qt5 and Qt6, right ?

closed 2025-05-08 by kadet1090