FreeCAD/FreeCAD #21205: DXF: Import of certain spline is not correct.

Roy-043, 2025-05-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

A certain spline from a DXF is not imported correctly. LibreCAD (DXF editor) and BricsCAD V18 (DWG editor) have no issue with the spline so it seems to be a correct DXF entity.

Forum topic:
https://forum.freecad.org/viewtopic.php?t=96720

1. Restart in safe mode.
2. Open the DXF from the ZIP file.
3. Do not change any DXF settings.
4. Result: there are 2 slot-shaped splines. The top one has an irregularity that is not present in LibreCAD or BricsCAD.
5. The Draft_Draft2Sketch command fails on that spline.

A check on the latest dev (on Ubuntu) will follow.

[spline-slots-2000.zip](https://github.com/user-attachments/files/20103570/spline-slots-2000.zip)

### Full version info

```shell
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/86.4394
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-05-08:
I tried on windows without safe mode.I can confirm. 

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * Curves 0.6.61
```

furgo16, 2025-05-08:
Is this the C++ or the Python importer?

ErtugrulEge, 2025-05-08:
> Is this the C++ or the Python importer?

also I have two warning messages when importing

```
20:57:07  Entity has zero-length extrusion direction
20:57:07  Entity has zero-length extrusion direction
```

[image]

[image]

Roy-043, 2025-05-09:
The test was done with the C++ importer. But the legacy importer does not import the spline correctly either.

Roy-043, 2025-05-09:
The "zero-length extrusion direction" error occurs for both splines. And checking the DXF shows that group code 210 is indeed 0,0,0 for both.
```
AcDbSpline
210
0
220
0
230
0
```

AndreiCherniaev, 2025-05-12:
FreeCAD 1.1.0 Revision: 41632 (Git) crash on this file. Is reason the same?

[freecad crash test.zip](https://github.com/user-attachments/files/20153099/freecad.crash.test.zip)

```
a@Linux:~$ freecad
FreeCAD 1.1.0, Libs: 1.1.0devR41632 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Adding snap-specific PYTHONPATH to sys.path: /home/a/snap/freecad/common/.local/lib/python3.10/site-packages:/snap/freecad/1447/lib/python3.10/site-packages:/snap/freecad/1447/usr/lib/python3/dist-packages
Gtk-Message: 11:20:59.057: Failed to load module "canberra-gtk-module"
Gtk-Message: 11:20:59.057: Failed to load module "canberra-gtk-module"
Qt: Session management error: Could not open network socket
connect failed: No such file or directory
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x74f9c0120520]
#1  0x74f9c0cce8d4 in QOpenGLContext::shareGroup() const from /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x4
#2  /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5(+0x500d35) [0x74f9c1041d35]
#3  0x74f9c1041f2f in QOpenGLFunctions::initializeOpenGLFunctions() from /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0xf
#4  /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5(+0x5036de) [0x74f9c10446de]
#5  0x74f9c1044b99 in QOpenGLFramebufferObject::QOpenGLFramebufferObject(QSize const&, QOpenGLFramebufferObjectFormat const&) from /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0xc9
#6  0x74f9c1044be8 in QOpenGLFramebufferObject::QOpenGLFramebufferObject(int, int, QOpenGLFramebufferObjectFormat const&) from /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x38
#7  0x74f9c357bb17 in Gui::Multisample::testSamples(int) const from /snap/freecad/1447/usr/lib/libFreeCADGui.so+0x67
#8  0x74f9c357bc18 in Gui::Multisample::supported() const from /snap/freecad/1447/usr/lib/libFreeCADGui.so+0x68
#9  0x74f9c3321041 in Gui::Dialog::DlgSettings3DViewImp::addAntiAliasing() from /snap/freecad/1447/usr/lib/libFreeCADGui.so+0x181
#10  0x74f9c33211dd in Gui::Dialog::DlgSettings3DViewImp::DlgSettings3DViewImp(QWidget*) from /snap/freecad/1447/usr/lib/libFreeCADGui.so+0x4d
#11  0x74f9c3128802 in Gui::PrefPageProducer<Gui::Dialog::DlgSettings3DViewImp>::Produce() const from /snap/freecad/1447/usr/lib/libFreeCADGui.so+0x22
#12  0x74f9c37032e8 in Gui::WidgetFactoryInst::createPreferencePage(char const*, QWidget*) const from /snap/freecad/1447/usr/lib/libFreeCADGui.so+0x18
#13  0x74f9c33182d8 in Gui::Dialog::DlgPreferencesImp::createPreferencePage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) from /snap/freecad/1447/usr/lib/libFreeCADGui.so+0x38
#14  0x74f9c331ad29 in Gui::Dialog::DlgPreferencesImp::createPageInGroup(Gui::Dialog::PreferencesPageItem*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) from /snap/freecad/1447/usr/lib/libFreeCADGui.so+0xb9
#15  0x74f9c331bf8f in Gui::Dialog::DlgPreferencesImp::setupPages() from /snap/freecad/1447/usr/lib/libFreeCADGui.so+0x5f
#16  0x74f9c331c061 in Gui::Dialog::DlgPreferencesImp::DlgPreferencesImp(QWidget*, QFlags<Qt::WindowType>) from /snap/freecad/1447/usr/lib/libFreeCADGui.so+0xa1
#17  0x74f9c307a92c in Gui::ApplicationPy::sShowPreferencesByName(_object*, _object*) from /snap/freecad/1447/usr/lib/libFreeCADGui.so+0x6c
#18  /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128008) [0x74f9c1a25008]
#19  /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x8c) [0x74f9c19dcadc]
#20  /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9dbc) [0x74f9c1978a1c]
#21  /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x74f9c1abf3af]
#22  /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x69de) [0x74f9c197563e]
#23  /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x74f9c1abf3af]
#24  /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x4b16) [0x74f9c1973776]
#25  /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x74f9c1abf3af]
#26  /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68) [0x74f9c19789c8]
#27  /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x74f9c1abf3af]
#28  /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe) [0x74f9c1aba3de]
#29  /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x20c4cd) [0x74f9c1b094cd]
#30  /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyRun_StringFlags+0x79) [0x74f9c1b0a4f9]
#31  0x74f9c207d22c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /snap/freecad/1447/usr/lib/libFreeCADBase.so+0x6c
#32  0x74f9c31702ac in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /snap/freecad/1447/usr/lib/libFreeCADGui.so+0x8c
#33  0x74f9c303f9e6 in Gui::Application::importFrom(char const*, char const*, char const*) from /snap/freecad/1447/usr/lib/libFreeCADGui.so+0x206
[... cut ...]
#58  0x74f9c0848d14 in QCoreApplication::exec() from /snap/freecad/1447/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#59  0x74f9c304941b in Gui::Application::runApplication() from /snap/freecad/1447/usr/lib/libFreeCADGui.so+0x76b
#60  /snap/freecad/1447/usr/bin/FreeCAD(+0x89bc) [0x589017beb9bc]
#61  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x74f9c0107d90]
#62  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x74f9c0107e40]
#63  /snap/freecad/1447/usr/bin/FreeCAD(+0x8cc5) [0x589017bebcc5]
```

ErtugrulEge, 2025-11-20:
> Thank you [@FEA-eng](https://github.com/FEA-eng) for diligently going through all these issues and testing whether they're still relevant!

but why we have to 190 days to resolve or make an activity  with an issue? I think issues shouldn't be automatically closed by github bot. Reporter or participants can't control  the issue again and bot can close the unresolved issue which is I think not the correct workflow.