FreeCAD/FreeCAD #21207: All Part Design sketch-based operations fail with invalid input error in the Windows weekly 2025.05.06

FEA-eng, 2025-05-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the current Windows weekly from the main repository, all PD sketch-based operations fail with the "Invalid input" error. To reproduce, create a Body, sketch a circle or rectangle and try to pad it.

Forum thread: https://forum.freecad.org/viewtopic.php?t=96869#p825869

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41585 (Git)
Build date: 2025/05/06 03:23:37
Build type: Release
Branch: main
Hash: 261ce705a93e98826389f20c4a99238757dfaab3
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-05-08:
See https://github.com/FreeCAD/FreeCAD/issues/20929#issuecomment-2822860431 concerning packaging types

maxwxyz, 2025-05-09:
@oursland FYI

69Piero, 2025-05-09:
this release is shockingly unusable and should be removed form repository

FEA-eng, 2025-05-11:
FEM is broken too - constraints cannot be applied to the geometry because of:

```
19:31:00  <Exception> TopoShapeExpansion.cpp(2019): Invalid shape name Face6
19:31:00  ConstraintFixed: Invalid shape name Face6
```

I guess it's the same issue. Reportedly, there are also problems with TechDraw.

maxwxyz, 2025-05-13:
It's not just with sketch-based operations, also just padding a face of a cube results in 

[image]