FreeCAD/FreeCAD #21212: cannot move annotation if created with frames off

jackbat1436, 2025-05-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Create a simple part. 
In techDraw,  Insert a page with views, put in a few measurements and turn off "View Frames"
Insert Annotation.

You should find that you cannot select or move the annotation without first enabling view frames. Once you have you can then disable them and everything seems to work fine.

When you insert the annotation, you should note that Lock Position is indeed false so it does not appear to be a false default setting.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Anything else?

I was able to repeat the issue 3 times in a row using the above steps and have included a screenshot of a very simple project that shows the steps I took.  Again, once you toggle the frames on and off, it appears to start working fine.

At first glance it appears the annotation is created with a default value of Lock Position : false, however you can not even select it by double clicking so the problem appears to be bigger than that. 

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-09:
Can you recreate this on 1.1dev ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets) 

Note: make sure to backup you 1.0 files before opening them in 1.1dev

WandererFan, 2025-08-14:
I believe that PR  "TechDraw: Rework view frames #22869" solves this problem.

closed 2025-08-16 by WandererFan