FreeCAD/FreeCAD #21216: Exported step has geometry issues while the source Body does not have them.

leoheck, 2025-05-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The exported step has geometry issues while the source Body does not have them.
Is this something known? Or is there a bug exporting step?

Here Body was analysed and does not have issues in Geometry.

[image]

But after exporting it to step (and importing it back), it has tons of problems.

[image]

```
  Face288 | Boolean operation: Invalid curve on surface
  Edge644 | Boolean operation: Invalid curve on surface
  Face289 | Boolean operation: Invalid curve on surface
  Edge645 | Boolean operation: Invalid curve on surface
  Face290 | Boolean operation: Invalid curve on surface
  Edge747 | Boolean operation: Invalid curve on surface
  Face291 | Boolean operation: Invalid curve on surface
  Edge751 | Boolean operation: Invalid curve on surface
  Face291 | Boolean operation: Invalid curve on surface
  Edge755 | Boolean operation: Invalid curve on surface
  Face291 | Boolean operation: Invalid curve on surface
  Edge679 | Boolean operation: Invalid curve on surface
  Face291 | Boolean operation: Invalid curve on surface
  Edge687 | Boolean operation: Invalid curve on surface
  Face291 | Boolean operation: Invalid curve on surface
  Edge681 | Boolean operation: Invalid curve on surface
  Face291 | Boolean operation: Invalid curve on surface
  Edge672 | Boolean operation: Invalid curve on surface
  Face291 | Boolean operation: Invalid curve on surface
  Edge653 | Boolean operation: Invalid curve on surface
  Face291 | Boolean operation: Invalid curve on surface
  Edge661 | Boolean operation: Invalid curve on surface
  Face291 | Boolean operation: Invalid curve on surface
  Edge655 | Boolean operation: Invalid curve on surface
  Face291 | Boolean operation: Invalid curve on surface
  Edge646 | Boolean operation: Invalid curve on surface
  Face291 | Boolean operation: Invalid curve on surface
  Edge758 | Boolean operation: Invalid curve on surface
  Face291 | Boolean operation: Invalid curve on surface
  Edge759 | Boolean operation: Invalid curve on surface
  Face291 | Boolean operation: Invalid curve on surface
  Edge760 | Boolean operation: Invalid curve on surface
  Face291 | Boolean operation: Invalid curve on surface
  Edge761 | Boolean operation: Invalid curve on surface
  Face291 | Boolean operation: Invalid curve on surface
  Edge763 | Boolean operation: Invalid curve on surface
  Face292 | Boolean operation: Invalid curve on surface
  Edge764 | Boolean operation: Invalid curve on surface
  Face292 | Boolean operation: Invalid curve on surface
  Edge665 | Boolean operation: Invalid curve on surface
  Face293 | Boolean operation: Invalid curve on surface
  Edge764 | Boolean operation: Invalid curve on surface
  Face294 | Boolean operation: Invalid curve on surface
  Edge779 | Boolean operation: Invalid curve on surface
  Face297 | Boolean operation: Invalid curve on surface
  Edge872 | Boolean operation: Invalid curve on surface
  Face303 | Boolean operation: Invalid curve on surface
  Edge873 | Boolean operation: Invalid curve on surface
  Face304 | Boolean operation: Invalid curve on surface
  Edge876 | Boolean operation: Invalid curve on surface
  Face305 | Boolean operation: Invalid curve on surface
  Edge880 | Boolean operation: Invalid curve on surface
[... 276 lines cut ...]
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.74.0
  * Curves 0.6.61
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.2.0
  * lattice2 1.0.0
  * kicadStepUpMod 11.6.9
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-05-08:
If the produced steps are not valid, I can't perform boolean operations, to compare between different versions.

maxwxyz, 2025-05-09:
please provide example files

leoheck, 2025-05-09:
Sorry, I can't.

I have a file with a body that generates a valid geometry, checked on every feature.
When exporting to step and importing back, it has geometry issues all over the place.

Is it a known bug?

leoheck, 2025-05-09:
I have many objects exported, no matter the complexity (I may find an example to share, just have to check...)

All of them report to me this:
```
Error in Face: BOPAlgo_InvalidCurveOnSurface
```

Some give this message:
```
Exception while processing file: ./check_geometry.py [cannot unpack non-iterable NoneType object]
```

I am checking the object this way:
```python
shape = obj.Shape
valid, errors = shape.check(True)

if valid:
    print(f"The shape of object '{obj_name}' is valid.")
else:
    print(f"The shape of object '{obj_name}' is NOT valid")
    if args.verbose:
        for err in errors:
            print(f"{err}")

```

When using `valid = shape.check(True)` instead of `valid = shape.check()`
It tells me the objects are valid.

leoheck, 2025-05-09:
Here is a sample file: [step_geometry_issues.FCStd.zip](https://github.com/user-attachments/files/20124169/step_geometry_issues.FCStd.zip)

The step was also imported into the file to make it easier for someone to check it.

Here are the results.
[image]

I am exporting the step this way:

[image]

Importing the step this way:

[image]

leoheck, 2025-05-09:
A hint of the problem may be here.
[geometry_issue_2.FCStd.zip](https://github.com/user-attachments/files/20128359/geometry_issue_2.FCStd.zip)

I could reproduce this where there is a draft angle and a pocket. 
However, I have files where I don't have draft angles. At least not using the Draft tool.
I may have draft angles on other features like Pocket, Pad, Hole..

maxwxyz, 2025-05-10:
@pieterhijma might this be OCC related?

luzpaz, 2025-05-10:
Any relationship to... ?
* https://github.com/FreeCAD/FreeCAD/issues/20588

leoheck, 2025-05-10:
I compared (using diff) the exported step of 0.21, 1.0 and 1.1.0devR41573 of the same Body (Body_Cylinder of my [geometry_issue_2.FCStd.zip](https://github.com/user-attachments/files/20128359/geometry_issue_2.FCStd.zip)) and they look exactly the same (except by the generation time)

The problem could be the import function or the geometry test.

leoheck, 2025-05-11:
Tested a different idea.

Exported the same Body - which Freecad does not see geometry issues - to igs.
Then found online something like this https://www.3dpea.com/en/convert/IGS-to-STP to convert to step.
Then when checking the geometry of the produced step file, it still has issues.

Also a different idea using MayaConv. I converted the step produced by Freecad, to another step. The content changed completely, but still has the same object. However this still produces a geometry issues.

UR-0, 2025-05-11:
Please try to Part->Compound->Set Tolerance to fix Check Geometry (workaround only!)

[image]

leoheck, 2025-05-11:
@UR-0 thanks, this is interesting.
I could also track the minimum tolerance until it starts to fail, which for this design was like 4.93 nm
[image]

Comparing the produced step with the original (with problems), only small portions change.

[image]

leoheck, 2025-05-11:
Ah, sad that if I export this solution to a step and import it back, it is still not good.

leoheck, 2025-05-12:
With my real design this Tolerance doesn't even work since the value is already high, like close to 1 mm, but then it reports me this

```
  File "/home/lheck/Applications/FreeCAD-41573/squashfs-root/usr/Mod/Part/BOPTools/ToleranceFeatures.py", line 129, in execute
    sh.fix(selfobj.minTolerance,selfobj.minTolerance,selfobj.maxTolerance)
<class 'RuntimeError'>: check failed, shape may be empty
21:07:59  Tolerance_new: check failed, shape may be empty
```

leoheck, 2025-05-12:
Even with a shape that is geometrically invalid, I can still make boolean operations. At least they worked on the simple example.

pieterhijma, 2025-05-14:
I've looked at it and on the latest FreeCAD with the latest OCCT I have no problems. The body passes the check and the imported step that was included in the file passes the check. I exported as step myself and reimported it again and that one also passes the check.

```
OS: Arch Linux (tty)
Architecture: x86_64
Version: 1.1.0dev.41697 (Git)
Build date: 2025/05/14 07:36:42
Build type: Unknown
Branch: main
Hash: 9d6f1ad37c1d97bcf669272c3124ce6045691b5b
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 6.9.0
shiboken 6.9.0, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 8.0.0.rc1-7aa85582ad
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
  * Curves 0.6.61
  * freecad.gears 1.3.0
```

leoheck, 2025-05-14:
I am a bit confused right now.

The weeklies from 1h ago, is 41694. 
Your version 41697 is in the future. 
Your build was done by yourself, right?

You are also using OCC `8`.
This is also newer, awesome.

Also, saw a different naming of weeklies with the datetime instead of the short version (not finding the page anymore, maybe it was just a quick transition).

Well, I am downloading this 41694 to check.

leoheck, 2025-05-14:
These versions still have the same issues.
I am assuming it may be due to OCC 7.8.1 then.

The Body does not have issues, but not the Exported/Imported Part

[image]

@pieterhijma any idea when we can have this version with OCC 8.0 available in AppImage?

```
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.41694 (Git)
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.74.0
  * Curves 0.6.61
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.2.0
  * lattice2 1.0.0
  * kicadStepUpMod 11.6.9
```

leoheck, 2025-05-14:
I found the other version again, it is in Freecad repo.
The short hash looks completely different.

Where do I get info on which one is the latest or the differences between these 2 versions?
Freecad is also building AppImages now?
Will Freecad-Bundle be discontinued?

```
➜ freecad_weekly_41694
Version: 1.1.0dev.41694 (Git) <================ 41694
Build date: 2025/05/13 12:46:54
```

```
➜ ./freecad_weekly_2025.05.13 
Version: 1.1.0dev.14555 (Git shallow) <========= 14555
Build date: 2025/05/13 09:39:01
```

pieterhijma, 2025-05-14:
> I am a bit confused right now.
> 
> The weeklies from 1h ago, is 41694. Your version 41697 is in the future. Your build was done by yourself, right?

Yes, ~~I enter this wormhole to see if it is fixed in the future~~ I test on the latest freecad (what is in the main branch) and I use the latest OCCT (also their main branch). If the problem does not occur on the latest version, I simply assume it is fixed and don't report it to them.
 
> You are also using OCC `8`. This is also newer, awesome.
> 
> Also, saw a different naming of weeklies with the datetime instead of the short version (not finding the page anymore, maybe it was just a quick transition).
> 
> Well, I am downloading this 41694 to check.

It is probably the OCC version that makes the difference.

pieterhijma, 2025-05-14:
> [@pieterhijma](https://github.com/pieterhijma) any idea when we can have this version with OCC 8.0 available in AppImage?

See [their blog post](https://dev.opencascade.org/content/important-announcement-occt-80-release-and-patch-migration-process). I think FreeCAD will follow soon.

pieterhijma, 2025-05-14:
> I found the other version again, it is in Freecad repo. The short hash looks completely different.
> 
> Where do I get info on which one is the latest or the differences between these 2 versions? Freecad is also building AppImages now? Will Freecad-Bundle be discontinued?

I don't know. @oursland, @adrianinsaval?

pieterhijma, 2025-05-14:
> > I exported as step myself and reimported it again and that one also passes the check.
> 
> [@pieterhijma](https://github.com/pieterhijma) do you mind sharing the exported step?

Sure.

[step_geometry_issues-Body.step.txt](https://github.com/user-attachments/files/20210527/step_geometry_issues-Body.step.txt)

leoheck, 2025-05-14:
I guess @adrianinsaval step files are broken. 
Step is the universal file for sharing the design with other CAD tools and vendors.

It has not been released yet, but since it is in RC1, we should start to see some movement to use it.

leoheck, 2025-05-14:
@pieterhijma, thanks for the file you shared.

Interesting, this is the visible added mass.
Compared to the one generated with OCC 7.8
[image]

And of course, tons of values changed inside.
[image]

leoheck, 2025-05-14:
Another interesting thing is that even with the step that was exported fine, FC 1.1 with OCC 7.8 cannot validate it as good.

oursland, 2025-05-14:
The FreeCAD repo built images are not fully functional on all platforms.  Consequently, the FreeCAD-Bundle, while outdated, is still preferred.

Upgrading OCCT is not possible on the `conda-forge` built images due to an outdated `smesh`.  Currently the most recent compatible OCCT release is 7.8.1.

I have outstanding questions regarding the appropriate path to upgrading `smesh`.  Notably, the current `smesh` both in `conda-forge` and in the 3rd party source within FreeCAD is based upon a very old release from SALOME with some patches applied.  Do we want to upgrade to the latest `smesh`?

adrianinsaval, 2025-05-14:
Upgrading smesh on conda-forge should be fine, on the FreeCAD copy it's not so simple because we need to keep support for a wider range of occt versions, 7.5.1 is our minimum but I think upstream only targets the current version of occt

adrianinsaval, 2025-05-14:
> A quick glance at the current `smesh` (9.14) looks like it's a substantial change to the CMakeLists.txt in which they depend upon `SalomeOpenCASCADE` instead of `OpenCASCADE` as before (7.7).

it's been like that since 2018 (or 8.5.0) it seems: https://github.com/SalomePlatform/smesh/commit/97889f80acb2a7e5d7c9aaa2b6c6b2557014b49d https://github.com/SalomePlatform/smesh/blame/5e46af59684e7cdf480e523e46c5e1b5dfc9ee12/CMakeLists.txt#L210

adrianinsaval, 2025-05-17:
is this issue solely based on the output of geometry check or is there a real issue when _using_ the step file for something? A failed geometry check only gives hints on where a problem _could_ be _if_ a boolean opration is failing, it should not be used as absolute guideline to judge a geometry

leoheck, 2025-05-17:
A problematic step file is being generated from valid bodies. It is observed from simple to complex desings. I could not find a single external tool that doesn't report issues on the step file. I could also not find a tool that can fix the issues. I am about to send step files to a company in china to produce them using injection molding which is an expensive process. With issues like those I cannot guarantee months of work will work fine on the tools they use. I also observed that many other issues were tagged as OCC related. Such upgrade is a must and may improve many aspects of the tool. Honestly it should receive the highest priority.

leoheck, 2025-05-17:
> I've looked at it and on the latest FreeCAD with the latest OCCT I have no problems. The body passes the check and the imported step that was included in the file passes the check. I exported as [...]

@pieterhijma did you do something special to be able to build using OCC 8.0-rc1?

adrianinsaval, 2025-05-17:
> I could not find a single external tool that doesn't report issues on the step file.

what external tools? what issues? I may try shipping 8.0-rc1 next week if this is a real problem but BOP check (specially if it's just invalid curve on surface) is not enough evidence IMO. BOP check often gives false positives, see the opinions of expert people on the forum: https://forum.freecad.org/viewtopic.php?style=5&p=669185#p669185 https://forum.freecad.org/viewtopic.php?p=204395#p204395 https://forum.freecad.org/viewtopic.php?p=686084#p686084 https://forum.freecad.org/viewtopic.php?p=669069#p669069

otherwise I'll get to it once 8.0 is realeased and begins the migration process on conda-forge, I will work eitherway on migrating to 7.9 at least though, maybe that fixes the issue too

leoheck, 2025-05-17:
what external tools? what issues? 
I found one or two step viewers online. Geometry issues. I don't know much about the problem unfortunately.

It is written somewhere on occ page that the biggest work is on 7.9 and the 8.0 is more like improving organization of the project. They also mention something about 8.0 being released next year if I am not mistaken. For me 7.9 would be enough if we could have step files that doesn't have geometric issues. this would also represent a big step to fix other occ related issues like problems with fillet and chamfers that happens all the time.

oursland, 2025-05-17:
With @adrianinsaval's changes to the `conda-forge` packages, we can now use `pixi` for the macOS and linux builds with OCCT 7.9.0 if we do not package `ifcopenshell`, but there is still a conflict on Windows due to uninstallable `pthreads-win32`.

leoheck, 2025-06-06:
Figured out the software the company uses is the Materialise Magics 2022 version (there are two releases this year, 26.0 and 26.01, but I am not sure which one it is).

Does anyone happen to have this software/version?