FreeCAD/FreeCAD #21218: Recovering after a crash with file with links IS A PAIN IN THE ASS.

leoheck, 2025-05-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

After a crash (this time due to a chamfer/fillet)
Fc proposes that I recover thousands of files that were not even opened.
I was just working with 2 or 3 files open.

I think that since they contain Assemblies (in this case using Asm4) with links, it wants to "restore" all the linked files, but they were not being edited at this time. This is annoying as hell!

Also, if I say yes, to restore the ones I was working on.  I then have to close all the other files that were not being edited, but now with fear, since they are in the "not saved" state... and since there are many it may happen that I close the one I want to have restored by accident.

Also, IF I save one of those "unsaved files", I receive this kind of complaint:

[image]

I can highlight 2 issues here.

1. There are 3 tabs (files open) BUT 2 of them are the SAME FILE.
2. If I try to save the filename called Body, it does not allow me with this screen. 

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.41573 (Git)
Build date: 2025/05/05 17:45:01
Build type: Release
Branch: main
Hash: 3189030c11010631e726dbe350e9077dc84109bb
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.74.0
  * Curves 0.6.61
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.2.0
  * lattice2 1.0.0
  * kicadStepUpMod 11.6.9
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-10:
Relevant:  
* #21104

FEA-eng, 2026-10-10:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20261001 (Git shallow)
Build date: 2026/10/01 12:05:00
Build type: Release
Branch: (HEAD detached at 99c5620c5)
Hash: 99c5620c5f739ca5c308849dda95ac468fec8b79
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 - Build 27.20.100.8854/Intel/Intel(R) UHD Graphics 620
```