FreeCAD/FreeCAD #21229: BIM: Arch_MergeWalls results in error when base objects are draft objects

semhustej, 2025-05-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Merging walls in the attached file results in error and need to restart FreeCAD. The problematic part is that base objects of walls are Draft_Lines. I am not able to reproduce the issue when the base objects are Sketches.

Steps to reproduce:

- open file (remove the .zip filename)

[072_merge_material-error.FCStd.zip](https://github.com/user-attachments/files/20125602/072_merge_material-error.FCStd.zip)

- select all three walls
- use `Utils` -> `Merge Walls` tool
- walls are merged, but one original wall remains in the file (which it should not) and error message shows up
18:20:06  Running the Python command 'Arch_MergeWalls' failed:
```
Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41365-conda-Windows-x86_64-py311\Mod\BIM\bimcommands\BimArchUtils.py", line 485, in Activated
    FreeCADGui.doCommand("Arch.joinWalls(FreeCADGui.Selection.getSelection(),delete=True)")
  File "<string>", line 1, in <module>
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41365-conda-Windows-x86_64-py311\Mod\BIM\Arch.py", line 1234, in joinWalls
    FreeCAD.ActiveDocument.removeObject(n)

Access violation
```

- FreeCAD needs to be restarted to work in a normal way after that

See behavior in the video:

https://github.com/user-attachments/assets/01df20af-7ad8-4be7-a2c9-c625f8b6d17d

I am NOT to reproduce the issue in FreeCAD 1.0. It looks like a regression.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * BIM021 2021.12.0
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-05-09:
@semhustej is your version of 1.0 based on Qt5.x, if so, then it's very likely this is basically the same as #20498, see my comment on the root cause analysis https://github.com/FreeCAD/FreeCAD/issues/20498#issuecomment-2764654545 ?

semhustej, 2025-05-09:
> [@semhustej](https://github.com/semhustej) is your version of 1.0 based on Qt5.x, if so, then it's very likely this is basically the same as [#20498](https://github.com/FreeCAD/FreeCAD/issues/20498), see my comment on the root cause analysis [#20498 (comment)](https://github.com/FreeCAD/FreeCAD/issues/20498#issuecomment-2764654545) ?

Yes, my FreeCAD 1.0 is Qt5 based:

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * BIM021 2021.12.0
  * Reporting
  * Telemetry 1.0.1

```

It is possible that Qt6 is the culprit.

But this reported issue might be different one, as it behaves in a different way than 20498:

- I can not reproduce it using Sketches as base objects. It is related to Draft_Line based walls. 
- It  does not crash FreeCAD, unlike 20498
- I can reproduce it 100% times, unlike 20498

If the issue gets confirmed, let's leave it open and test once https://github.com/FreeCAD/FreeCAD/issues/20498 gets resolved.

Syres916, 2025-05-10:
> It is possible that Qt6 is the culprit.

Let's say the API in Qt6 is now subtly different from Qt5 so most scripts that work on Qt6 also work on Qt5 but a lot less work the other way around. When we're using Python to interact with the `QTreeWidget` we are using PySide and it's the Tree that's the issue here, it's not specific to Merging Walls or even BIM objects.

Just to help reinforce the point, I've been using a (very slightly modified) version of _ToggleVis_ macro on Qt5 based FreeCAD builds since I helped test it on 0.19 (primarily uses the `QTreeWidget` for performance all on Qt5 based builds). I thought the changes to get it working on Qt6 would be very minor, it turns out it was way beyond my capabilities and only vaguely resembles the original code. See https://github.com/Syres916/FreeCAD/commit/3c3f4bbd6e6411fd91788b924891b92e1806a47c

Roy-043, 2025-06-17:
> it's the Tree that's the issue here

I have to admit I was surprised by this, but testing with the Model panel disabled shows that the problem then does not occur. So a relation with the Tree is definitely there.

Is there anything that can be done on the Python side to catch this error?

Syres916, 2025-06-17:
@Roy-043 this is now fixed after #21965 was merged, tested using:

```
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.42189 (Git)
Build date: 2025/06/17 13:59:00
Build type: Release
Branch: main
Hash: e17a83fa1fb64e8449a7d645ea6725020c934100
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 6.9.0
shiboken 6.9.0, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.9124/1
Installed mods: 
  * MeshRemodel 1.10.37
  * FreeCAD-themes 2025.1.7
  * Assembly4 0.50.18
  * CfdOF 1.30.4
  * Woodworking 1.0.1.39285.1.0.1.2025.06.04
  * sheetmetal 0.7.22
  * fasteners 0.5.38
```

Roy-043, 2025-06-18:
Thanks for that information. Closing this then.

closed 2025-06-18 by Roy-043