FreeCAD/FreeCAD #21231: CI: Lint checks are malfunctioning

luzpaz, 2025-05-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Testing for tabs, whitespace returns:  
```
Run python3 tools/lint/generic_checks.py \
  File "/home/runner/work/FreeCAD/FreeCAD/tools/lint/generic_checks.py", line 39
    error_lines = [idx if line.endswith(" ") for idx, line in enumerate(lines, start=1)]
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
SyntaxError: expected 'else' after 'if' expression
Error: Process completed with exit code 1.
```

[image]

ref: https://github.com/FreeCAD/FreeCAD/actions/runs/14927312886/job/41962047014?pr=21221

### Full version info

```shell
n/a
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

benj5378, 2025-05-10:
Maybe it is `idx for idx, line in enumerate(lines, start=1) if line.endswith(" ")` then

benj5378, 2025-05-10:
```python
>>> a = [" asd", "asd2 ", "ad32 ", "345"]
>>> [x if x.endswith(" ") for x in a]
  File "<stdin>", line 1
    [x if x.endswith(" ") for x in a]
     ^^^^^^^^^^^^^^^^^^^^
SyntaxError: expected 'else' after 'if' expression
>>> [x for x in a if x.endswith(" ")]
['asd2 ', 'ad32 ']
```

closed 2025-05-13 by hyarion