FreeCAD/FreeCAD #21235: Varsets are not properties in real life

dfcautley, 2025-05-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have recently converted a large number of spreadsheets to varsets, so I've been spending a lot of time in varset land.

1. I agree that renaming a varset is important. From reading it seems like we're headed in that direction.
2. Some of my varset lists are big. Finding something to reference is troublesome.

The sort order of varsets doesn't make sense to me. It seems to be alphabetic in some cases, but not all. I am probably not helping myself because my spreadsheet -> varset macro sorted all the aliases before creating the varset. At any rate, if I create a new varset property via context menu, the new property is not inserted in sort order. In fact closing and re-opening the file doesn't change this.

[image]

In this example the property 'fop' was added and is displayed out of order.

And now, to the actual subject line.

**Varsets are very different from the properties of an object.**

I am very interested in the idea that varsets and spreadsheet cells are both stored as properties under the covers.
While this is very attractive from a code/method re-use point of view, I think it's a problem to treat everything that is stored as a property the same. I don't want users renaming or sorting standard object properties, but custom properties and varset values are different critters.

I don't have a solution for this (sorry)... but I find it interesting and hope that we don't make the UI/UX the same for all of the sub-classes of properties. In other words, I suggest that the workflow should be different for each sub-class of properties. It already is vastly different for spreadsheets.

just 2c.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/74.6389/1.25
Installed mods: 
  * BillOfMaterials 1.0.2
  * boltsfc 2022.11.5
  * Cables 0.1.4
  * Curves 0.6.61
  * DynamicData 2.74.0
  * fasteners 0.5.38
  * Help 1.0.3
  * Manipulator 1.6.0
  * ose-piping
  * parts_library
  * Quetzal 1.3.3
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-05-10:
The sorting of properties is case-sensitive.

FlachyJoe, 2025-05-11:
Varset doesn't pretend to be a property, it's a property container object.
And Varset properties are same as any object properties.
[image]