FreeCAD/FreeCAD #21236: BIM: Arch_Axis Tool When the axis is rotated by 90 degrees, the mouse zoom stops working

kaiwas, 2025-05-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

How to reproduce the problem.
1) Add an axis
2) Go to the axis editing and change the angle to any other than zero.

After this, the mouse zoom stops working.

Reopening the file does not solve the problem.

Video demonstration.

https://github.com/user-attachments/assets/8a11e157-7c1c-4395-ba4f-1396c3a6fa76

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41628 (Git) AppImage
Build date: 2025/05/09 10:50:32
Build type: Release
Branch: main
Hash: 9577d108e333c80ee640211271a76105f52baf14
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_SketchArch-master.zip
  * FreeCAD_SketchArch-master
  * Road 2025.2.22
  * PitchedRoof.zip
  * TrackingGroups
  * Curves 0.6.61
  * freecad.gears 1.3.0
  * TrackingGroups.zip
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-05-10:
I think this issue is same as : https://github.com/freecad/freecad/issues/19578

Other issues I reported about arc axis
https://github.com/freecad/freecad/issues/19620
https://github.com/freecad/freecad/issues/19618
https://github.com/freecad/freecad/issues/19615
https://github.com/freecad/freecad/issues/19579

Please check also those issues

kaiwas, 2025-05-10:
> I think this issue is same as : [#19578](https://github.com/FreeCAD/FreeCAD/issues/19578)

Thank you!
A similar problem indeed. Probably I didn't use the right phrases to search for similar problems.

closed 2025-05-11 by maxwxyz