FreeCAD/FreeCAD #21237: Preference "Default unit system" not saved

wsteffe, 2025-05-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Using Preference Editor I may set a different "Default unit system" in the current session but after a restart the  unit system goes back to Internal(m,..).

I do not know if it is related but, upon starting FreeCAD I get the message:
```
During initialization the error "UI file does not exist" occurred in /usr/local/Mod/BIM/InitGui.py
Please look into the log file for further information
Preferences file :/ui/preferences-sh3d-import.ui not found
```

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (GNOME-Classic:GNOME/gnome-classic/xcb)
Architecture: x86_64
Version: 1.1.0dev.41613 (Git)
Build date: 2025/05/07 21:22:49
Build type: Unknown
Branch: main
Hash: 3b47a4ca4a154541553802d73c29b828a2262e08
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.7.0
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/141.585/1
Installed mods: 
  * asm3 0.12.0
  * Assembly4 0.50.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-05-10:
It's most likely fixed when https://github.com/FreeCAD/FreeCAD/pull/21174 is merged but I'm also testing a follow-up PR to put the default description and units back to how they were.

Syres916, 2025-05-10:
Having done some investigation, it's actually a different regression from #20540 caused by the creation of the Units combobox in the Start module and the setting of the current index, the index keeps being set to `0` which changes the `user.cfg` and therefore propagates through the entire app. Hopefully I'll have a fix by tomorrow.

Syres916, 2025-05-10:
Having looked at the old setup pre-PR20540, the trace should be something like:

```
MainWindow void retranslateUi()
MainWindow unitChanged ()
Start GeneralSettingsWidget::retranslateUi()
```

but in current main the `Start GeneralSettingsWidget::retranslateUi()` isn't being triggered until much later by which time the index has been changed to 0 but I cannot figure out where and I'm getting out of my depth. @hyarion a fresh set of eyes maybe needed here.

Syres916, 2025-05-12:
I think I've worked out a fix for the Start module messing up the user's set schema and it's included in https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Revert_Internal_metres_to_Standard_millimetres so I've just a few extra checks to add to the TechDraw dimension test which should be done in the next couple of hours.

hyarion, 2025-05-12:
@Syres916 note there's a refactoring going on by @bofdahof and @3x380V https://github.com/FreeCAD/FreeCAD/pull/21147

PR #21466 (2025-05-20): Start: fix restoring of user saved Unit schema
Fixes: #21237
closed 2025-05-20 by hyarion