FreeCAD/FreeCAD #21240: Remove new PartDesign boringtool.

berberic2, 2025-05-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Remove the new PartDesign boringtool: it is totally broken for weeks and unmaintained.

- Values of head-types value are not applied correctly to boring-parameters, making head-type-settings unusable 
- Order of drop-down-boxes is wrong. head-type is populated by standard, so standad shplud be above head-type

This changes were obviously not tested at all and the contributor does not care.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41632 +2 (Git)
Build date: 2025/05/10 06:38:58
Build type: Release
Branch: Meins
Hash: a32e91953cb1b632c6a78dcdb66157fd9478c1c0
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.8
shiboken 5.15.8, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.6.3
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/87.9548/1
Installed mods: 
  * Assembly4 0.50.18
  * sheetmetal 0.7.22
  * Silk 0.1.7
  * DynamicData 2.74.0
  * ExplodedAssembly
  * Curves 0.6.61
  * Pyramids-and-Polyhedrons
  * freecad.gears 1.3.0
  * fasteners 0.5.24
  * A2plus 0.4.68
  * Render 2024.12.15
  * CurvedShapes 1.0.13
  * QuickMeasure 2022.10.28
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-05-10:
I have edited your error report and removed the snarky bits. If you do not have the patience for the dev version you should not use it.

berberic2, 2025-05-10:
> I have edited your error report and removed the snarky bits. If you do not have the patience for the dev version you should not use it.

I object.

I have used the dev-version for years. For years it was possible to use it and so find bugs, problems, generate bugreports or even commit a fix. As oss-projects typically have no testing-department, this  is the way the development-process of oss-projects maintain quality. 
But since some time, every time I compiled dev, it was unusabel, in a different way.

If no one is using the dev-version, because it is unusable → no testing
If no testing → no bugreports, no fixes
If no bugreports and no fixing → bugs stay

The development-process of freecad is broken.

berberic2, 2025-05-10:
Having an issue with the title _this is an issue_ is total useless and carries no information at all.

berberic2, 2025-05-10:
For confirmation see:

https://forum.freecad.org/viewtopic.php?t=96633#p823968
https://forum.freecad.org/viewtopic.php?t=96633#p824838

hyarion, 2025-05-10:
It is not unmaintained, you said yourself that it was recently changed. You can’t have it both ways.

There’s a lot of change at the moment as we’re in the middle of a dev cycle. Feedback is welcome but keep it constructive.

It’s good to point out issues but that doesn’t mean it needs to be reverted, instead it should be fixed.

hyarion, 2025-05-10:
Locking issue for a while

closed 2025-05-10 by hyarion

berberic2, 2025-05-20:
FYIO: After this incident I decided to stop contributing to FreeCAD development.