FreeCAD/FreeCAD #21241: FreeCAD Dev 1.1.0 CAM Not able to create an adaptiive tool path (1.1.0 geom issue)

ShamanTcler, 2025-05-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Creating a simple tray, with a pocket made from an adaptive tool path.

When the model is made with 1.1.0 dev adaptive path just hangs. Using the same 1.1.0 dev model I can use the pocket operation just fine. The 1.1.0dev model uses an offset sketch to create the wall of the model. (single offset value constraint)

Dropped back to 1.0 release model recreated the model, this time single offset constraint not available so a bunch of dimensions. Here adaptive and pocket both work in 1.0. Now I open the 1.0 model in 1.1.0 dev ... this time adaptive works just fine.

The file convention is:

BasicTray,FCStd            1.1.0dev
BasicTray1_0.FCStd       1.0 release

[BasicTrays.zip](https://github.com/user-attachments/files/20140124/BasicTrays.zip)

### Full version info

```shell
OS: Windows 11 build 26120
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windows11
Logical DPI/Physical DPI/Pixel Ratio: 96/101.976/1
Installed mods: 
  * A2plus 0.4.68
  * btl 0.9.9
  * Curves 0.6.62
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-10:
Just wondering, can you confirm none of these issues are preexisting? https://github.com/FreeCAD/FreeCAD/issues?q=state%3Aopen%20label%3A%22Mod%3A%20CAM%22%20adaptive

ShamanTcler, 2025-05-10:
I checked before submitting, and even though there are a lot with adaptive none felt right. Incidentally there are no splines in the model.

Everything behaves as expected to begin with… select face, adjust a few params and hit apply. You get the spinning wheel that never finishes.

Creating the model in 1.0 rel and the tool path is created correctly. 

Just now I took the 1.1.0dev model, changed it back to LinuxCNC post processor, and loaded it into 1.0 Release. Tried to create the adaptive toolpath and it would not create.

dbtayl, 2025-05-15:
Stock is set as the endmill, which has no overlap with your model, so nothing gets cut:
[image]

Change that to what I assume was intended to be the stock and it works fine:
[image]

Stock is set correctly in BasicTray1_0, which is why that one worked.

ShamanTcler, 2025-05-15:
Thank you... I will check it out. 

I am curious what led you to find this issue? I would like to learn more about.

dbtayl, 2025-05-15:
I don't have a good answer to that. From my experience it seemed unlikely that the adaptive operation would fail like you describe, so I looked for other explanations and found that the stock was set up incorrectly. Nothing particularly insightful, I'm afraid.

luzpaz, 2025-05-17:
@ShamanTcler please report back your results, TIA!

ShamanTcler, 2025-05-19:
Rolling on the floor laughing .... talk about stupid.  To keep it short yes, it works fine.

Now for the background so this makes sense.

This project is intended to cut small trays out of random chunks of wood. To do so everything must be parametric, and the CAM module is just not that smart when it comes to updates.

All the parameters are stored in a spreadsheet, so they get passed cleanly. For the stock I actually create a solid called stock from parameters. Then in the CAM job I select the "stock" from an existing solid called stock. At least that is what I intended, until I selected a random solid, in this case the end mill.

The name "stock-*" will now hopefully become trigger for me to remember. It was not obvious before.