FreeCAD/FreeCAD #21242: PartDesign: No planar face for sketch on side of tapered extruded rectangle

depthoffocus, 2025-05-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Extruding a simple rectangle in Part Design with a heavy taper, and then trying to attach a sketch to one of the extruded sides, brings up the error message: "You need a planar face as support for a sketch!"

Shallower tapers succeed.

The equivalent attachment mechanism in the Part workbench will successfully attach the sketch, but there may be something wrong with the geometry as Sketcher sees it -- I'm going to try to raise a separate issue for that.

This problem exists in 1.1-dev-41628 *and also in 1.0*.

https://github.com/user-attachments/assets/7a483ec0-f8d7-4e73-8478-d76f6e67e196

[no-planar-face.zip](https://github.com/user-attachments/files/20141211/no-planar-face.zip)

(rename attached to no-planar-face.FCStd)

### Full version info

```shell
OS: macOS 15.4.1
Architecture: x86_64
Version: 1.1.0dev.41628 (Git)
Build date: 2025/05/09 10:50:32
Build type: Release
Branch: main
Hash: 9577d108e333c80ee640211271a76105f52baf14
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.75/2
Installed mods: 
  * ThreadProfile 1.94.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.22
  * Curves 0.6.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Katt-Kushol, 2025-05-11:
Confirmed. An angle up to 34 degrees work. More than that does not. 

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41214 (Git) Conda

depthoffocus, 2025-05-11:
The related case I was going to raise as another ticket is probably better detailed here because I think it sheds light on it.

If you set the taper to a lower number, say 30 degrees, attach the sketch, it succeeds.

Then import one tapered edge as external geometry, draw a line and try to set them parallel. 

You get told you cannot, because "one selected edge is not a valid line".

If you move that edge and instead try to set them perpendicular, you get a series of error messages and among them get told "perpendicular constraints between line and bspline are not supported".

https://github.com/user-attachments/assets/5e79737a-4363-46d0-9e32-c64b5fa4f4c1

(If you set the taper to zero and try again, these operations succeed, but if you change the taper without deleting the sketch first, you can trigger a TNP issue.)