FreeCAD/FreeCAD #21245: Base: new Internal unit system duplicates Standard unit system, replaces Standard

furgo16, 2025-05-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

After the merge of #20540 a new unit system called `Internal` was introduced. This new unit system, at least from a user point of view, is exactly the same as `Meter decimal`. More important than the duplication, is also the removal of the `Standard` unit system.

Before:

[image]

After:

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41632 (Git) Snap 1445
Build date: 2025/05/09 19:27:21
Build type: Release
Branch: main
Hash: ce368cb7a0dd6a3f57e5cabc3e7745c5669683a9
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Telemetry 1.0.1
  * Assembly4 0.50.17
  * Launcher
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-05-11:
@furgo16 the order of the schemas is dealt with in PR #21174 specifically https://github.com/FreeCAD/FreeCAD/pull/21174/files#diff-74013290b8d19b8381ae674f0b4d4eb2a6016582d3610f99fafeb2c7fa039729 and I hope this gets a high priority.

With regard to the default schema (num=0) I've got a branch https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Revert_Internal_metres_to_Standard_millimetres which puts both the name and base unit back to how they were previously but most importantly adds Unit Tests that actually fail, I'm still adding to these. Once PR21174 is merged I should be finished and will submit a PR immediately thereafter. The PR20540 passed all tests even though three serious regressions were introduced and for me that's not acceptable.

The other regression that I'm really struggling with is the fact that the user cannot change the schema of their choice and it be retained on restart as reported in https://github.com/FreeCAD/FreeCAD/issues/21237 and to me at least the issue is the Start module Units combobox but I maybe wrong and would explain why I'm struggling.

furgo16, 2025-05-11:
Thank you.

> and I hope this gets a high priority.

Indeed, I feel all issues related to the regressions introduced with #20540 are disruptive and numerous enough to grant the Blocker status label:
- #21172
- #21245
- #21237
- #21209

@maxwxyz FYI

Syres916, 2025-05-12:
With regard to the Start module issue I was struggling with, progress has been made see https://github.com/FreeCAD/FreeCAD/issues/21237#issuecomment-2872316404

Syres916, 2025-05-12:
@FEA-eng as part of https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Revert_Internal_metres_to_Standard_millimetres which will become a PR later today hopefully, please be aware that I'm trying to make the core Unit Schema names consistent across the app so if someone searches for `ImperialCivil` they will find _all_ references to it. If you want the `FEM` putting back to `FemMilliMeterNewton` then I'll change the C++ data on https://github.com/FreeCAD/FreeCAD/blob/main/src/Base/UnitsSchemasData.h#L567 to match

3x380V, 2025-05-13:
@furgo16, could you verify #21174 is fixing issues you listed?

furgo16, 2025-05-13:
@3x380V happy to test, but before starting a ~2h build, I'd like to understand whether that PR is intended to fix all of them or whether other changes need to be applied. Currently:
- Of the [4 issues listed here](https://github.com/FreeCAD/FreeCAD/issues/21245#issuecomment-2869612025) only one of them has a linked PR
- I know @Syres916 was working on some additional fixes, but I've lost track. Have they been added to https://github.com/FreeCAD/FreeCAD/pull/21174 or do some additional fixes need to be applied on top?

3x380V, 2025-05-13:
@furgo16, I cherry-picked @Syres916's fix into that PR, so it should fix all the issues.

obelisk79, 2025-05-13:
the regressions are rather significant

closed 2025-05-14 by hyarion

FEA-eng, 2025-05-15:
@marioalexis84 Please note this: https://github.com/FreeCAD/FreeCAD/issues/21245#issuecomment-2872429937

PR #21174 (2025-05-14): Base: fix Unit scheme management regressions
Fixes: #21172
Fixes: #21209
Fixes: #21245
Fixes: #21316
Fixes: #21317