FreeCAD/FreeCAD #21246: Enhancement: Provide a menu icon/keyboard shortcut for the 'skip recompute' feature

herrdeh, 2025-05-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The 'skip recompute' checkbox in the model tree is a really useful feature when dealing with more complex models. Unluckily,
1. it's quite hidden, so very much unknown
2. it does not come in so handy, needs too much clicking for a really often used feature
3. a menu icon/keyboard shortcut cannot (tell me that I'm wrong if I am) be added, as there is no "skip recompute" entry in [menu] > tools > customize > toolbars 

To create this entry mentioned under 3. would perfectly solve the problem and would speed up work with very complex models dramatically. 

Thanks a lot in advance, 
Wolf

PS: #20907 closed, as this suggestion seems to be the much better solution. 

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * 3D_Printing_Tools
  * Assembly4 0.50.12
  * dodo 1.0.1
  * ExtremeProLight 1.6.0
  * Curves 0.6.29
  * BOLTSFC
  * nurbs
  * animation
  * IconThemes
  * lattice2 1.0.0
  * ProDarkThemePreferencePack 1.0.0 (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

B0cho, 2025-05-15:
Hi @herrdeh,
Could you please provide example _complex_ file, which would be nice to test the feature?
Can you also point where currently you can skip recompute, as you describe in point 1?
I would try to implement 3.

herrdeh, 2025-05-16:
/1: Sorry, my complex models are 1000s of hours of work, I cannot publish them here. I can send one to you privately, if that helps.
/2: Root of model (very top item in the model tree) ==> right click ==>  checkbox "skip recompute" (sorry, screenshots do not work with a right click object.

Thank you very much for taking care of this, much appreciated!

luzpaz, 2025-05-28:
> About icon: Can I create a icon pixmap for `Skip recompute`? Or are Icons done by art contrubutors?

ping @FreeCAD/design-working-group

herrdeh, 2025-06-16:
Thank you so much for implementing this. Can't wait having it available!

Cheers, Wolf

herrdeh, 2025-08-20:
Found it in recent daily builds, but still without graphics.
It's sooo useful, makes the Freecad experience so much smoother and more convenient. Hope this feature will make it into default visibility, so people learn about it. 
Interesting question: How do we advertise new features? - To be honest: I don't read changelogs... ~)
 
Thank you so much for implementing this, I owe you a beer.
W