FreeCAD/FreeCAD #21248: BIM: Wrong placement for newly creatted Arch Window and  BIM Door

zisco, 2025-05-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Adding a Arch Window or a BIM Door on a Wall result in the creation of the object but in the wrong placement. The newly created object is placed at (0,0,0) rather than the point on the chosen face of the wall.

Step to reproduce:
1) open BIM workbench
2) create a line on XY plane
3) create a wall using the line as base object
4) select a face a of the wall
5) change the working plane to the selected face
6) place a window (or a door) on a point 

### Full version info

```shell
OS: Fedora Linux 42 (Workstation Edition) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.41628 (Git) AppImage
Build date: 2025/05/09 10:50:32
Build type: Release
Branch: main
Hash: 9577d108e333c80ee640211271a76105f52baf14
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/161.962/1
Installed mods: 
  * dodo 1.0.1
  * OpticsWorkbench 1.2.0
  * sheetmetal 0.7.22
  * Reinforcement
  * Curves 0.6.61
  * FreeCAD_SketchArch
  * freecad.optics_design_workbench 0.6.2
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-05-11:
Hi,

it is hard to reproduce without seeing video. I am not able to reproduce.

Also, the window in the screenshot does not seem to be placed at (0,0,0)

One note:
- there is no need to change the working plane. As per wiki information: https://wiki.freecad.org/Arch_Window#Usage
`In the [3D view](https://wiki.freecad.org/3D_view), move the window to the location where you wish to place it. If you move the pointer over an [Arch Wall](https://wiki.freecad.org/Arch_Wall), the outline of the window should align itself with the face of that object.`

zisco, 2025-05-11:
https://github.com/user-attachments/assets/95b390b0-172c-4f70-b6d2-67e522f4ba49

kaiwas, 2025-05-11:
This ArchSketch

semhustej, 2025-05-11:
@paullee0 should this issue be closed as duplicate of https://github.com/paullee0/FreeCAD_SketchArch/issues/27 ?

kaiwas, 2025-05-11:
> [@paullee0](https://github.com/paullee0) should this issue be closed as duplicate of [paullee0/FreeCAD_SketchArch#27](https://github.com/paullee0/FreeCAD_SketchArch/issues/27) ?

The problem is solved, but the change has not been merged yet. We just need to wait a little bit.

paullee0, 2025-05-11:
As @kaiwas indicates, there was a PR which fixes the regression reported in this issue and improve UI/UX consistency pending to be reviewed and merged.

https://github.com/FreeCAD/FreeCAD/pull/21088

paullee0, 2025-05-13:
Fixed by this PR merged in FC:

https://github.com/FreeCAD/FreeCAD/pull/21088#event-17631759163

This could be closed.

closed 2025-05-14 by luzpaz