FreeCAD/FreeCAD #21250: Sketcher: Move/Array multiple copies, then application hangs

edwardpas, 2025-05-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Created a square with round corners 6.0x6.0x1.2mm. Move/Array 11 copies 8.0mm 0 deg. - ALL GOOD
Select all 12 Squares (All elements for squares in Elements), Move/Array 11 copies 8.0mm 90 deg - Application hangs

### Full version info

```shell
OS: Windows 11 build 22621
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Namibia (en_NA)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * 3D_Printing_Tools
  * addFC 1.5.1
  * ArchTextures
  * boltsfc 2022.11.5
  * Cables 0.1.4
  * CadbaseLibrary 2.0.0
  * sheetmetal 0.7.22
  * WB_Organizer 2024.1.29
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-11:
Please share an example file.
Can you try if the issue also occurs in 1.1dev?

luzpaz, 2025-05-11:
@edwardpas you can d/l the weekly (1.1-dev) from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets  

Just make sure to backup any 1.0 models before opening them in 1.1-dev

Lgt2x, 2026-01-18:
> What's the diagnosis here ?

We're working with a large constraint matrix (about 6k^2), and many operations in the sketcher solving routine process each matrix value individually. https://github.com/FreeCAD/FreeCAD/pull/26977 is a mitigation, speeding up the most time-consuming function, but the operation is still very long. There may be other spots to optimize, but there is an incompressible amount of operations to do to solve sketcher constraints.