FreeCAD/FreeCAD #21258: CAM: Job editor crashes / UI elements missing

maxwxyz, 2025-05-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

From: https://github.com/FreeCAD/FreeCAD/pull/20307#issuecomment-2851439318

CAM Job editor crashes randomly in various places with similar PySide errors like `RuntimeError: Internal C++ object (PySide.QtGui.QWidget) already deleted.` or random UI objects missing attributes, etc.
https://www.tech-artists.org/t/pyside-in-maya-internal-c-object-already-deleted-for-returned-pyside-objects/4877/6

@phaseloop suspects 
Firstly, unnecessary recalculation. Many ops will fully recalculate on closing even when nothing has changed or if Cancel Btn is pressed. There is very little tracking of "dirty" states. The presumably also impacts when variable changes trigger updates.

Second is the idea that error checking is not "good python style". There seems to be the idea that instead of checking whether something exists before referencing it, you just do it anyway and let it raise and exception if it fails. Then the exceptions are not trapped where they occur (that's bad practice too) and are allowed to bubble up and cause a mess later, maybe with a chain reaction of uninitialised objects.

### Full version info

```shell
tbd
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

knipknap, 2025-05-18:
The reason that bubbling is usually recommended in Python is that Python provides really good tracebacks, even for nested exceptions, and with a full trace identifying bugs is just easier. Hence bubbling is fully intended and 90% of the time it is the best error handling you can do in Python.

The example mentioned *could* (and I don't see a root cause definitely identified) mean that a traceback was not handled well *on the interface to C++* (e.g. in an event handler from C++), which is a specific sub case. The best fix would there be to fix the root cause - the traceback not being handled well at the interface - rather than removing bubbling from all Python code.

It's a classic case of leakage - the best thing is always to fix the leak, rather than to fight the symptom or avoid getting close to the pipe. Enforcing explicit catching of exceptions seems like not the right thing to do IMO. We can't turn Python into C++ because a problem may have leaked through the interface.

I am aware that a true fix would likely have to be done in PySide though. If the above theory even proves right. I think stuff like that can also happen if something happens in the wrong event loop, but not sure if and where CAM uses threading or async.

sliptonic, 2026-09-22:
Referenced issue is closed and we don't have a reproducible case here.  closing since no action is possible.

closed 2026-09-22 by sliptonic