FreeCAD/FreeCAD #21259: Esc key in Sketcher problem

wadysal, 2025-05-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Annoying operation of the Esc key in Sketcher

### Full version info

```shell
It would be very helpful if pressing the Esc key repeatedly in the Sketcher would **NOT EXIT the Sketcher**, but would remain in the editor. I should only be able to exit the Sketcher when I select "Exit." When designing, I often press Esc repeatedly to exit a function, but NOT TO EXIT THE EDITOR.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-05-11:
There is an option in Preferences to disable using `Esc` to leave the sketcher. This might solve your issue.

[image]

I would prefer to have `Esc` disabled to exit sketcher by default too, but I am not a sketcher power-user.

luzpaz, 2025-05-11:
Related:  
* https://github.com/FreeCAD/FreeCAD/issues/18944

luzpaz, 2025-05-11:
@wadysal please always add your full [About](https://wiki.freecad.org/About) info to tickets so we know what version you're reporting about.

closed 2025-05-12 by maxwxyz