FreeCAD/FreeCAD #21262: QT6 migration broke the Classic theme on Windows

FEA-eng, 2025-05-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Since the migration to QT6, the Classic theme on Windows is pretty much unusable, mainly because of that:

[image]

but it also looks like a strange mix with Light theme and has too large menu texts among others.

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41585 (Git)
Build date: 2025/05/06 03:23:37
Build type: Release
Branch: main
Hash: 261ce705a93e98826389f20c4a99238757dfaab3
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windows11
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-12:
it was broken with Qt5 too, that's why it was removed from the first start setup and automatically the light theme was set.
@MisterMakerNL FYI.

FEA-eng, 2025-05-12:
> it was broken with Qt5 too, that's why it was removed from the first start setup and automatically the light theme was set.

Well, I had no such issues before migration to Qt6. Maybe Classic should be removed completely or fixed.

maxwxyz, 2025-05-12:
@FreeCAD/design-working-group what to do with Classic Theme on macOS and Windows?

obelisk79, 2025-05-12:
I had suggested we enforce fusion as the QStyle for cross platform consistency a while back, if changing the qstyle resolved the issues reported above then I think we need to revisit that conversation.

@FEA-eng does executing this from the python console resolve the issues?
`QtWidgets.QApplication.setStyle('Fusion')`

obelisk79, 2025-05-12:
My mistake... do this first and then it should work:
`from PySide import QtWidgets`

FEA-eng, 2025-05-12:
It indeed helped - eliminated the issues mentioned here. Some aspects of the Light theme still remained but I guess they could disappear after restarting FreeCAD if this command remained active.

obelisk79, 2025-05-12:
It should be persistent, but you can verify by checking in the parameter editor here:

[image]

FEA-eng, 2025-05-12:
I had to add it manually, it wasn't added by that command.

PR #21379 (2025-05-26): Gui: Set QStyle to Fusion if not set
setting the qstyle to fusion fixes various margins and issues across OSs, it seems that the Basic Style of Qt6 is not good enough

here is a comparison under Debian Trixie using Qt6
| Basic | Fusion |
|-|-|
| [image] | [image] |

this should fix #21262