FreeCAD/FreeCAD #21263: Transform tool not working. 1.1

GAZ082, 2025-05-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When I import an SVG, do the things to get it into my Body as a binder and try to do a transformation, the result rolls back as soon as i click OK.

Tested this with simple bodies, including more than ones solid bodies and works fine, but when i work with logos in SVGs, fail.

https://github.com/user-attachments/assets/fdcca423-bb02-4842-936a-db0cdbb34d5c

[LEGO Freecad.FCStd.zip](https://github.com/user-attachments/files/20150528/LEGO.Freecad.FCStd.zip) (remove the .zip).

### Full version info

```shell
OS: Arch Linux (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.41628 (Git) AppImage
Build date: 2025/05/09 10:50:32
Build type: Release
Branch: main
Hash: 9577d108e333c80ee640211271a76105f52baf14
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/101.6/1
Installed mods: 
  * fasteners 0.5.38
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-11:
When did this regression show up ? (It's a regression, right?)

GAZ082, 2025-05-11:
Just started using 1.1 a couple of weeks ago and this always happened with imported SVGs. Tried with multi-shaped svgs, singe shaped svgs, all the same.

mnesarco, 2025-05-11:
@GAZ082 if you put the object inside a Part Container does it work?

Katt-Kushol, 2025-05-11:
Behavior confirmed.

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41214 (Git) Conda
Build date: 2025/04/09 18:28:54

[image]

GAZ082, 2025-05-12:
> [@GAZ082](https://github.com/GAZ082) if you put the object inside a Part Container does it work?

Yes it does, but i lose the ability to use the center of mass.

maxwxyz, 2025-05-12:
Can someone report the placement of the body before and after the transformation? I suspect that the body moves, but the binder keeps it's original location (there is a property for that in the binder). I cannot check it myself as latest Windows builds are broken.
You can also enable Show placement = true in the view tab of the body to show it's origin.

GAZ082, 2025-05-12:
@maxwxyz hey Max:

This is the LEGO body. It changed.
BEFORE:
[image]
AFTER:

[image]

closed 2025-05-12 by maxwxyz

GAZ082, 2025-05-17:
Worked, thanks!